tzapu/WiFiManager #170: Multiple problems with WiFiManager

WilbertE, 2016-05-13:
Hi guys,

First of all, nice library. But I'm having some troubles.

1. The first time I connected to the chip, a login portal shows up on my iPhone automaticly. Now a couple days later, I can connect to the chip but the portal never shows up.

2. When I set my browser to go to the IP address (192.168.4.1) I do get to the pages which normally are in the portal. But when I go to the page with all the networks I sometimes get half sended pages with some abrupt endings. It's like parts of the page are not send and some html is missing.

Any tips on this?

kentaylor, 2016-05-14:
I'm guessing what you mean here but if you are saying it works on 192.168.4.1 but not other addresses then it might well be because you are using https. The browser does not allow redirection for https addresses.

For issue 2. I had this problem when I clicked on the Configure WiFi button but not when clicking on the Configure WiFi (No Scan) button. Is that the case for you, in which case I can suggest a fix, or is it something else?

WilbertE, 2016-05-14:
I will try to explain it better.

First of all, I edited the webpages the ESP8266 sends ([WiFiManger.h, lines 25 to 36](https://github.com/tzapu/WiFiManager/blob/master/WiFiManager.h)), so I have my own logo (base64 image) for my project. (can this create the problems by making the data to send a little longer?) ([this is my webpage code](http://pastebin.com/YMFp2ksr))

So to explain things better I created the video. I resetted the ESP8266 with the following command:
`wifiManager.resetSettings();
      delay(3000);
      ESP.reset();
      delay(5000);`. It did erase the settings and it showed up as AP again. I named it "Aquacontroller" becouse its part of an fish tank project.

[This is the video](https://www.youtube.com/watch?v=8uEBxKvslas)
It's dutch, but you will understand.

1. First you see me try to connect to the ESP8266 but the portal never shows up automaticly.
2. Then I try to go to the IP address but becouse the iPhone detects there is  no internet it has switched to my normal WiFi again. (I try this two times)
3. Then I uncheck the "Automatically log in" for the wifi, and it stays connected to the ESP8266 and doesn't auto jump to other Wifi.
4. I then surf to the IP address but uncompleted pages are being send.

tzapu, 2016-05-14:
ah, cool, you sorted it :)

i was about to say that a forget network during development does wonders, ios decides at times (and you found the exact scenario when, on cancel) to not show the portal...

it looks very damn good, i must say

good luck

kentaylor, 2016-05-14:
I don't fully understand what you say. Apple says "Tapping Cancel on the Log In screen disassociates the device from the captive Wi-Fi network" but you never did that in the video so why do you say "If the portal is canceled on an iOS device it wont show up anymore".

Second I've tested WiFiManager with a few users and found lots of confusion. These videos are a great way to see exactly what happened and some of the confusion it causes. You always need new users for testing because once a user knows what is going on they adjust their mental model and no longer get confused. Once you know, you can never see the world like a new user does. Only through watching virgin users can you learn what they see.

In your first video you have the Auto-Login switch on. Apple says "Your iPhone, iPad, or iPod touch will remember the network and login information so you can automatically reconnect to that network when you're in range". So your Iphone is going through a sequence of requests to try and join the internet. That means in the case of WiFiManager the Iphone is sending a sequence of requests to WiFiManager telling it to do goodness knows what, I guess it depends on the sequence you used in the past. 

Webservers generally complete requests in parallel, opening a new thread for each http request. The ESP webserver queues them and completes them sequentially. The http response to a request must be completed before the next one is commenced. If that sequence includes a WiFi scan request the device can take quite a long time to complete the scan before it will respond. If the browser sends a scan request more than once the ESP device will queue the requests and complete the scans sequentially. The browsing device will abandon the earlier requests and only wait for a response from the most recent. That can take a long time as all of the WiFi scans are completed sequentially so the browsing device times out.

So in the first video when you first connect to the network and the browser fails to download a page that is because there are requests that haven't finished completion from the Iphon'e Auto-Login logic and the browser request is queued after them. The Auto-Login logic fails and that is why it switches back to the normal WiFi network. Then you reconnect to the ESP device network and in your browser you get the root page. That is lucky, both the browser and Auto-Login are firing requests at WiFiManager and the sequence they arrive there depends on timings from each. From here I'm not sure of the sequence because at some point the Auto-Login logic fails again and switches you back to the normal WiFi network but that happens in the background and you can't tell in the video when it occurs while you are interacting with the browser. What I do see is that you request a WiFi scan at https://youtu.be/8uEBxKvslas?t=46s and 2 seconds later do it again. I don't understand why you do not see the waiting page after the first click and you see it after the second but I think you have now requested two scans. Perhaps you do not see it after the first click because it is still loading the GIF image between the first and second click. At some point, before you get a response to the second wifi scan request you get switched off the ESP network. 

You then turn the Iphone's Auto-Login off at https://youtu.be/8uEBxKvslas?t=59s . Here you connect but you do not wait for what Apple calls the "Welcome screen" before switching to the browser. You now get the root page, request the scan once and get an incomplete response. The scan takes a long time so I suspect the browser timed out before it got the complete response or maybe the Iphone switched back to the normal network before it got the complete response. This may occur because you didn't go to the Welcome screen yet and if you don't go there, it times out and switches back to the normal network. I'm not sure of this because I don't know the Iphone logic for switching networks and in the video you do not go back again to see what network you are on.  When you tried to request a scan again you were probably already on the normal WiFi network so got no response. You then requested the root page and saw the image but this came from the Iphone cache and the ESP web server never saw the request.

In the second video you never use the Iphone browser but interact with the ESP from the "Welcome screen" so expect different behaviour. It is not obvious from your explanation that you were aware of that. I've only tested with Android but I have certainly seen users configuring the ESP from the portal configuration screen (the android equivalent of the Welcome screen) while thinking they were using the normal browser. In this case the Iphone closes the Welcome screen when the ESP device switches off the WiFi network. If you had been using a browser you would only have seen the web page failing to load when the network switched off with no indication why.

You concluded "Large webpages results in problems. " I doubt that was a problem and I'm not sure the GIF animation was a problem either. 

I've done a version of WiFiManager that among other things keeps the web server running after you have configured a network so that you get web pages that tell you of your success or otherwise. I think this workflow is less confusing. It would be great if you give it a try and provide some feedback. See https://github.com/kentaylor/WiFiManager/

WilbertE, 2016-05-14:
> I don't fully understand what you say. Apple says "Tapping Cancel on the Log In screen disassociates the device from the captive Wi-Fi network" but you never did that in the video so why do you say "If the portal is canceled on an iOS device it wont show up anymore".

I didn't do it in the video, it happend before. I came to this github to find information why it happend that the portal didn't show up anymore. When I couldn't find information I tried to find it out myself.
So first I wanted to know if it was the ESP8266 or the iPhone what was causing the portal not showing up. I tried to connect to the AP with my iPad and it did show up. So it had to do something with the iPhone. When I searched the internet to get more information about how portals worked I came to the Apple site pointing out "_Tapping Cancel on the Log In screen disassociates the device from the captive Wi-Fi network_". And I could remember I did tap the cancel button the day before to try connecting by IP address. So I thought, that must have been the reason why it doesn't show up on the iPhone but does show up on my iPad.

Because you ask more information I tried to replicate the problem and I could do it. And [offcourse recorded it in this video](https://www.youtube.com/watch?v=6ukz_bAwpkI) ;-). 
I'm being a little inpatient in the video becouse I didn't want the video to become to long. But you can see what is happening. 
So the portal doesn't show up anymore when:
- You connect with an iOS device to the AP. 
- Visit 1 page or more in the portal
- Tap the "cancel" button and choose "keep connected without internet".

> Second I've tested WiFiManager with a few users and found lots of confusion. These videos are a great way to see exactly what happened and some of the confusion it causes. You always need new users [...]

Glad I can help with that ;-). If I can give one tip: use loading screens. The first time I used the library I tapped the "Configure WiFi" button and thought that nothing happend or I mistapped so I tapped again. But in the background the chip was scanning for networks. In my own code I added a JavaScript loading screen, as you can see in the video's. That way there is always feedback to the user.

> Webservers generally complete requests in parallel, opening a new thread for each http request. The ESP webserver queues them and completes them sequentially. The http response to a request must be [...]

Well I guess that would be fixed by a loading screen so a user cannot tap twice on the same button. There won't go multiple request to the ESP and the browser won't timeout, right?

> https://youtu.be/8uEBxKvslas?t=46s and 2 seconds later do it again. I don't understand why you do not see the waiting page after the first click and you see it after the second but I think you have [...]

No, the first tap is mis tapped. The second one is a good tap. You can see by the loading bar in the top that the first tap isn't registered. The GIF isn't loading when I tap the button, the gif was already on that page but had a `display: none` style to it. When I tap the button some javascript makes it visible. No loading screens are being loaded after a tap, they already hidden on that same page.

> You then turn the Iphone's Auto-Login off at https://youtu.be/8uEBxKvslas?t=59s . Here you connect but you do not wait for what Apple calls the "Welcome screen" before switching to the browser. You [...]

No that didn't happen. What happens was as follows: The portal didn't show up, so I had to go to the IP manually so I could configure the wifi. But when I started Safari the iPhone swiched to a Wifi network with internet acces and disconnected from the ESP, [@20+ sec.](https://www.youtube.com/watch?v=8uEBxKvslas&feature=youtu.be&t=20s)). At 40 seconds I try it again but same thing but again the iPhone swiched networks.
What happend: When I connected to the ESP the iPhone tried to autologin. But off course it did not succeed. The portal would not show up anymore because I pressed cancel the day before. So It decided the WiFi was worthless and switched to a good network. And I cannot reach the IP.

At [58 seconds](https://www.youtube.com/watch?v=8uEBxKvslas&feature=youtu.be&t=58s) I disable autologin. What happens: When I connect to the ESP, the iPhone does not try to autologin. And because it doesn't, it doesn't switch networks because it does not check if it is a good network by auto login. This way I can reach the IP manually, I'm still connected to the ESP.

> In the second video you never use the Iphone browser but interact with the ESP from the "Welcome screen" so expect different behaviour. It is not obvious from your explanation that you were aware of [...]

The second video I had resetted the Wifi by "forgot network" and reboot the iPhone.This way the portal does show up again. As you can see in the video at the top of this post.

If you have more questions or need more video's don't hesitate to ask.

kentaylor, 2016-05-15:
You are going to a lot of trouble to follow up on this which is nice.

> So the portal doesn't show up anymore when:
You connect with an iOS device to the AP.
Visit 1 page or more in the portal
Tap the "cancel" button and choose "keep connected without internet".

Avoiding failure modes is fairly complicated for someone configuring an ESP device for the first time and they also have to have Auto-Login off except for the first time they access that particular SSID. In the first video you do all the configuration through the browser but it always switches to another network before you complete. There ought to be a way to make it stay on the ESP network so that browser configuration works. In the Android 5.1 version I've tested and Chromebook there is no Auto-Login functionality. The work flow is  different, it doesn't automatically pop up a config portal screen, and people generally did the configuration from the browser when I watched them. I thought ESP WiFi configuration was non intuitive in Android and Chromebook but Apple looks even more complicated. I expect you want it to work from any browsing device. I do and I wondered how it looked on Apple.

> In my own code I added a JavaScript loading screen, as you can see in the video's. That way there is always feedback to the user...Well I guess that would be fixed by a loading screen so a user cannot tap twice on the same button. There won't go multiple request to the ESP and the browser won't timeout, right?
 
Yes that is nice. I'd like the version at https://github.com/tzapu/WiFiManager/ to have something similar.

Can you tell from the browser screen when the WiFi network switches? 

> it doesn't switch networks because it does not check if it is a good network by auto login.

Are you sure it doesn't switch networks some time after [58 seconds](https://youtu.be/8uEBxKvslas?t=58s)?

> I'm just trying to find reasons why things happen. 

As am I and it is easier with your videos. I've found configuring the WiFi network through a browser to be far more difficult and complicated than I expected.

closed 2018-01-26 by tablatronix