tzapu/WiFiManager #172: AutoConnectWithFSParameters issue with #include ordering

vrolijkegast, 2016-05-16:
I was unable to compile the AutoConnectWithFSParameters example

The error message was:
<ArduinoJson-path>/ .... / Polyfills/isNaN.hpp:29:17:  error: 'isnan' was not declared in this scope
    return isnan(x);

After investigation, I found out that there was a conflict between <WiFiManager.h> and <ArduinoJson.h>. I looks like WiFiManager does an 'undef' of isnan. Maybe it somehow includes cmath.h ? When I reordered the include statements, it compiles without issues. I stopped investigating after that.
The code now looks like this:

// this needs to be included BEFORE <WiFiManager.h>
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

Since it took me quite some time, I thought I should share it.
I am using ArduinoIDE 1.6.8. on Ubuntu Linux.
I compile the code for a 'Wemos D1 mini' (based on the ESP8266)

kentaylor, 2016-05-16:
That suggests a [feather in the cap ](https://goo.gl/FygtbL) of @battika for getting the order this way in his version at https://github.com/kentaylor/WiFiManager/blob/master/examples/ConfigOnSwitchFS/ConfigOnSwitchFS.ino

tzapu, 2016-06-02:
hi, just to let you know that the undefs and everything related to that has been reverted in the latest version

thanks for reporting and using ;)
reopen if still an issue

closed 2016-06-02 by tzapu