tzapu/WiFiManager #176: Question: Arduino uno with esp8266 (not standalone)

pewh, 2016-05-23:
Hi, this is a question rather than an issue.

Can I use this library with Arduino UNO + esp8266 shield? I don't have usb to serial, so I don't wanna use esp8266 as standalone. Thanks.

tzapu, 2016-05-23:
hi, unfortunately now. this is only for the standalone

cheers

closed 2016-05-23 by tzapu

pewh, 2016-05-23:
Thanks for your quick response.

There are many usb to ttl in the store, like FT232RL (red color, more expensive) and PL2303 (blue color, cheaper). Shall this library compatible with PL2303?

tzapu, 2016-05-24:
hi, the library does not depend in any way on the usb to ttl you use. if you can flash the esp with it, it should be good

cheers

oliviermichaud1, 2016-11-11:
Hi , 
I thanks you for your code , working nicely !    .I use it with nodemcu v3  . Now I need more analog PIN  . So I would like to use it with Arduino UNO+ESP8286-01 module .

Is it possible ?

tzapu, 2016-11-11:
hi, no, it would be a completely different environment

cheers

oliviermichaud1, 2016-11-11:
Hi , 

Thank you very for your quick answer . So i will use a multiplexer then i can get signal from multiple analog sensors . 

thanks .
,

levarish, 2017-08-27:
Hi,
I face similar challenge in my setup:
The  Arduino Uno runs the main program and have multiple  sensors and system logic.  ( including onewire, ~5 I/Os etc. )
The ESP8266 - runs the WiFi manager - works great, as ESP8266 standalone

Any suggestions how to integrate the Uno and ESP8266 + WIFImanager?
 - the WIFImanager is really important for the project to be free of any  hardcoding ....
Thanks