tzapu/WiFiManager #179: Unable to access page on HTTP server after few AP mode cycles

Zyxmn, 2016-05-31:
Hi, I am unable to access the config html page after a couple AP cycles. Here is what I did to test this:

1) I set the config portal timeout to 3 minutes and the device enters into a permanent loop of AP cycles . I use the autoconnect() function. I enter the wrong password for the chosen ssid after scan. 

2) The device tries to connect with the wrong password and fails and goes back to AP mode. After timeout it starts another fresh AP mode by calling the autoconnect() function again.  This is where I am unable to access the html page to configure the device. Even though the last message I get on the serial monitor is the  " *WM: HTTP server started " message.

I did notice that during AP modes where I am unable to load the html page, there are no
" *WM: 204 No Response ",
" *WM: Request redirected to captive portal " and
" *WM: Handle root "

messages on the serial monitor.

tzapu, 2016-06-02:
hi, i ve seen some issues similar to this when the module flash size was not set properly in the arduino ide

can you check that that is ok?

Zyxmn, 2016-06-03:
Hi @tzapu , the flash size setting is 4M(1M SPIFFS)

lkeehon, 2016-08-01:
Hi @tzapu , I'm facing similar problem based on the scenario below:
wifiManager.setAPCallback(configModeCallback);
1)  I'm running my program and wifi was turn off thus automatically went to AP mode
2) Able to connect to the AP but can't go into the setting page http://192.168.4.1

However, I can go into the setting page if i reset my board. I have tried numerous time but same result. Appreciate your kind advice and thanks in advance for your kind help

Note: I did access to the AP before 3 minutes timeout.
**Code**
WiFiManager wifiManager;
    
    wifiManager.setAPCallback(configModeCallback);
    wifiManager.setConfigPortalTimeout(180);
  
  if (!wifiManager.autoConnect(hostString)) {
    Serial.println("failed to connect and hit timeout");

    ESP.reset();
    delay(1000);
  }

closed 2018-01-26 by tablatronix