tzapu/WiFiManager #180: Crash on AutoConnectWithFSParameters

odilonafonso, 2016-06-02:
I tried to use this sample, without changes, got this error:

```
mounting FS...
mounted file system
reading config file
opened config file
{"serverName":"api.example.com","accessToken":"128du9as8du12eoue8da98h123ueh9h98"}
parsed json

Exception (28):
epc1=0x4000bf3c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff0220 end: 3fff0520 offset: 01a0

>>>stack>>>
3fff03c0:  3fff0544 3fff1c80 3ffef43c 40206adb  
3fff03d0:  3ffe8fa8 00000000 000003e8 feefeffe  
3fff03e0:  3fff129c 3fff0de4 feefeffe feefeffe  
3fff03f0:  feefeffe feefeffe feefeffe feefeffe  
3fff0400:  feefeffe feefeffe feefeffe feefeffe  
3fff0410:  feefeffe feefeffe feefeffe feefeffe  
3fff0420:  feefeffe feefeffe feefeffe feefeffe  
3fff0430:  feefeffe feefeffe feefeffe feefeffe  
3fff0440:  feefeffe feefeffe feefeffe feefeffe  
3fff0450:  feefeffe feefeffe feefeffe feefeffe  
3fff0460:  feefeffe feefeffe feefeffe feefeffe  
3fff0470:  feefeffe feefeffe feefeffe 3ffe87e8  
3fff0480:  feefeffe 3fff1c74 00000200 feefeffe  
3fff0490:  feefeffe feefeffe feefeffe feefeffe  
3fff04a0:  feefeffe feefeffe feefeffe feefeffe  
3fff04b0:  feefeffe feefeffe feefeffe feefeffe  
3fff04c0:  feefeffe feefeffe feefeffe feefeffe  
3fff04d0:  feefeffe feefeffe feefeffe feefeffe  
3fff04e0:  feefeffe feefeffe feefeffe feefeffe  
3fff04f0:  feefeffe feefeffe feefeffe 3ffef4f8  
3fff0500:  3fffdad0 00000000 3ffef4f0 4020e3b4  
3fff0510:  feefeffe feefeffe 3ffef500 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,2)

 ets Jan  8 2013,rst cause:4, boot mode:(1,2)

wdt reset
```
Any suggestion ?
Thanks

odilonafonso, 2016-06-03:
After formatting, without crash

closed 2016-06-03 by odilonafonso

DavidTruyens, 2017-07-30:
Hi @odilonafonso, @tzapu,

I have the same issue. And it works again after formatting. But if I push the reset button with formatting turned off it crashes again. 

I'm using an Adafruit Huzzah with the HTTPRedirect from @electronicsguy.

nhaduy99, 2017-07-31:
Hi, I had the same problem. I have a solution that:
This is due to the FS parameter problem at the beginning after resetting the ESp8266, the WifimanagerFSparameter.ino file firstly will read the JSON file which has already saved in FS memory. When you fist program to a new ESP, FS file has no JSON file in its FS memory, then it causes error like that.
To solve that problem I did following steps:
Step 1: You have to reprograme again but commented out the 4 lines (In my program I added 4 parameters and the ESP will check and read them in FS at the beginning, other parts is good to let it be) We commented out to let the ESP ignore it to prevent the error.
[image]

Step 2: Then you will get no error any more, get into the WiFi Configuration and add values for 4 parameters.  Check if the FS is saved all parameter or not. But this time when you reset the ESP it will not read the 4 parameters because you have commented the 4 lines above.

Step 3: After ESP saved the parameter successfully, then you un-commented these 4 lines again and upload the code 1 more time to make the ESP read the FS at the beginning.
Then next time you will get no error until you want to add new parameters (or change the name of parameters)
That's it, the problem solved for me. How about you, please comment is issue your result.

Regards
Seppy

DavidTruyens, 2017-07-31:
Hi @nhaduy99 ,

seems to work! I'll do some more testing soon.

Thanks for your reaction!