tzapu/WiFiManager #184: Strange error when booting.

kommancs96, 2016-06-20:
When I try to boot my device I get an exception before WM connects. Yesterday it worked well, however today I can not boot up my device. I made sure, the device gets enough power.  This is the output to the serial:
```
*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff07c0 end: 3fff0c30 offset: 01a0

>>>stack>>>
3fff0960:  3fff23b8 00000001 0000000d ffffff80  
3fff0970:  40236f9e 4021fad8 00000018 3fff1e88  
3fff0980:  00000001 3fff2350 3fff2350 40237184  
3fff0990:  3fff2350 3fff1e88 3ffefac3 00000000  
3fff09a0:  3ffef004 3ffeeea8 1271094c 3ffefac3  
3fff09b0:  40243308 00000064 f934fe1a fffeffff  
3fff09c0:  00000018 0000ffff 00000000 00000001  
3fff09d0:  3fff09e8 40233740 3ffeed94 3fff1e88  
3fff09e0:  f934fe1a 3fff5e2b 00000000 00000000  
3fff09f0:  00000000 00000000 00000000 00000000  
3fff0a00:  00000000 00000000 00000000 00000000  
3fff0a10:  00000000 00000000 40234437 3fff1e88  
3fff0a20:  3ffefac3 3fff1e88 3fff1e88 4023447f  
3fff0a30:  0000000c 3ffeeea8 3ffe9cbc 00000000  
3fff0a40:  40218909 00000003 00000003 3ffefac3  
3fff0a50:  40218a21 00000003 00000003 4020ca69  
3fff0a60:  00000001 40218a7a 00000003 3ffefac3  
3fff0a70:  40206cb0 3ffefac3 3ffe86e3 3ffefac3  
3fff0a80:  40206cde 3ffe86e3 3ffefac3 40206d04  
3fff0a90:  00000003 00000000 3fff0b30 4020ad11  
3fff0aa0:  3fff1ec8 0000000f 3fff0b30 4020b9c7  
3fff0ab0:  40206cc0 3ffefac3 3fff0af0 4020e6ae  
3fff0ac0:  40107158 3fff0b30 3fff0af0 4020e6fc  
3fff0ad0:  3ffe9001 4024229d 3fff0c80 00000000  
3fff0ae0:  00000000 3ffe86e3 3fff0b30 4020bb49  
3fff0af0:  3fff1ea8 0000000f 00000000 3fff1e88  
3fff0b00:  0000000f 00000000 3fff0c80 4020a3fe  
3fff0b10:  0001c200 3fff0c80 3fff0b30 3ffefc04  
3fff0b20:  3fffdc20 3fff0c80 3ffe86e3 40206998  
3fff0b30:  00000000 00000000 3ffe8ea4 00000000  
3fff0b40:  3fff1e48 0000000f 00000000 3fff1e68  
3fff0b50:  0000000f 00000000 00000000 00000000  
3fff0b60:  00000000 3ffe92c0 00000000 3ffe92c0  
3fff0b70:  00000000 3ffe92c0 00000000 3ffe92c0  
3fff0b80:  00000000 3ffe92c0 00000000 3ffe92c0  
3fff0b90:  00000000 00000000 ffffffff 00000001  
3fff0ba0:  3ffe9001 00000000 00010035 402068f4  
3fff0bb0:  00000000 00000000 00000000 00000000  
3fff0bc0:  00000000 00000000 00000000 00000000  
3fff0bd0:  00000000 00000000 00000000 00000000  
3fff0be0:  00000000 00000000 00000000 00000000  
3fff0bf0:  00000000 00000000 00000000 00000000  
3fff0c00:  00000000 00000000 00000000 00000000  
3fff0c10:  00000000 00000000 3ffefbfc 4020c096  
3fff0c20:  00000000 00000000 3ffefc10 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,3)
```

kommancs96, 2016-06-20:
Here is my setup code:
```
//Main setup
void setup() {
  // put your setup code here, to run once:
  Serial.begin(BAUD_RATE);

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  DEBUGLN("Trying to connect to WiFi...");
  WiFiManager wifiManager;

  wifiManager.resetSettings();

  //set callback that gets called when connecting to previous WiFi fails, and enters Access Point mode
  wifiManager.setAPCallback(configModeCallback);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect(UUID)) {
    DEBUGLN("failed to connect and hit timeout");
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(1000);
  }

  //if you get here you have connected to the WiFi
  DEBUGLN("Connected to WiFi");

  
  return;
}
```

tueddy, 2016-06-21:
Hello, 

same here on booting:

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff0bc0 end: 3fff0ff0 offset: 01a0

I'm sure this was working before. Tried version 0.11.0 & 0.12.0.
Tried also format the filesystem.

Best regards
Dirk

kommancs96, 2016-06-21:
The AutoConnect example also crashes with the same exception. I found that the code possibly crashes at [this line](https://github.com/tzapu/WiFiManager/blob/master/WiFiManager.cpp#L246).

tzapu, 2016-06-21:
hi, could you confirm what esp8266 core lib you are using and what arduino ide?
also, did you set the flash size to the proper size in arduino ide?

kommancs96, 2016-06-21:
Hi! I think I use version 1.0.0 of the esp8266 library. I install the compiled binary using esptool.py in Terminal.

tzapu, 2016-06-23:
hi, try and update to the latest esp8266 core please 
https://github.com/esp8266/Arduino#installing-with-boards-manager

or at least to 2.2.0

cheers

tueddy, 2016-06-24:
Hello, same problem after updating.

Ardunio IDE 1.6.9
ESP8266 package: 2.3.0
Wifi-Manager:0.12.0

tested with different new boards: Witty Cloud, Wimos.

tzapu, 2016-06-25:
hi, thanks for letting me know.
i did not test with 2.3.0 myself yet. does this happen on a simple AutoConnect example?
i'm quite swamped currently, but hope to get a bit more time next week and hopefully attend to this
sorry for any inconvenience

kommancs96, 2016-06-25:
Yes, it happens with AutoConnect example. But on my chip the WiFiScan example works, so the wiring and power supply has to be correct.

kentaylor, 2016-06-30:
You could have corrupted flash. Try erasing the flash with this sketch https://github.com/kentaylor/EraseEsp8266Flash .

closed 2018-01-26 by tablatronix