tzapu/WiFiManager #186: unable to set access point name

romuye123, 2016-06-22:
Following is my code:

```
 WiFiManager wifiManager;
 wifiManager.resetSettings();
 wifiManager.autoConnect("My Chip name");
```
Even if i set access point name it displays "ESP_xxxxxx" in the wifi list.
What is wrong with this ?

romuye123, 2016-06-24:
@tzapu : waiting for your response

ahmadmn, 2016-06-24:
after
WiFiManager wifiManager;
Append this code:
WiFi.hostname("YourName");

romuye123, 2016-06-24:
@ahmadmn: Thanks for your quickly reply.  Tried appending the code, but it still does not work. 
Here is my updated code:
`WiFiManager wifiManager; 
WiFi.hostname("mychipname"); 
wifiManager.resetSettings(); 
wifiManager.autoConnect("testchipname");` 

Here is debug output: 
`0
chksum 0x0f
csum 0x0f
*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.
*WM: 
*WM: AutoConnect
*WM: Two
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: SET AP STA
*WM: Three
*WM: testchipname
*WM: 
*WM: Configuring access point... 
*WM: testchipname
*WM: AP Name Here
*WM: testchipname
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started`

Following is the screenshot on my phone:

[image]

tzapu, 2016-06-24:
hi,

that is the default username of the esp chip, somehow it is not getting set by wifimanager at all...

can you updated to the latest everything (esp8266 core, wifimanger) and try again? also make sure you are setting the correct flash size for your esp module

cheers

ahmadmn, 2016-06-24:
It works for me.
I use Arduino 1.6.9 and following libs:
[image]

romuye123, 2016-08-02:
Works fine after updating to latest

closed 2016-08-02 by romuye123