tzapu/WiFiManager #187: Using Websockets after Autoconnect

kommancs96, 2016-06-23:
Hi !
I recently made a working example where I am able to switch a LED using HTTP protocol.
The problem is that sending http requests, waiting for response, than process the response is too much time. I am curious if I am able to use websockets for communication after successfully connecting to WiFi using WiFiManager?

ahmadmn, 2016-06-24:
Hi
I need Async compatible WiFiManager too.

tzapu, 2016-06-24:
hi, i am using websockets and wifimanager in a project here, hope it helps
https://github.com/tzapu/WebSocketSerialMonitor/blob/master/WebSocketSerialMonitor.ino

async webserver i have not used yet, but i guess it needs some intialisation/reset/reintialisation after wifimanger is done

cheers

ahmadmn, 2016-06-25:
Thanks @tzapu
The example works fine.
But I have a basic question!
I want to write a Delphi application to communicate with ESP8266 (WebSocketSerialMonitor.ino included).
I must write a WebSocket server or client?

tzapu, 2016-06-25:
in my particular app linked above, the esp8266 is a websockets server.
but you could have it be either server or client depending on what you need it to do, look at the websockets lib docs
cheers

kommancs96, 2016-06-25:
Thanks, @tzapu , helped a lot. ;)

ahmadmn, 2016-06-26:
Thanks @tzapu,
Your code works perfect with delphi:

[image]
[image]

rooky06, 2017-04-13:
hello **ahmadmn**
I downloaded  IPwors activeX trial but theres only demo for VB and ASP.
Can you share your code to i learn( i m newbie)
thank you

ahmadmn, 2017-04-22:
Hi rooky06

rooky06, 2017-04-23:
Hello

Thank you for the link.

I downloaded it .

I go to try to use it like the sample code here 
https://developers.binary.com/demos/

thank you

closed 2018-01-26 by tablatronix