tzapu/WiFiManager #192: Using On Demand Portal - set default SSID and KEY?

Humancell, 2016-07-07:
I"m using the On Demand Portal, and I'm curious if there is a way that I can default the SSID and Key to be values that I provide, or even the currently set values?

I'm using the portal to configure numerous settings, and so each time I enter the portal I'm forced to re-select the SSID, and re-enter the Key each time.

I want to be able to default these to the current settings, and in some case default them to be specific initial values for testing.

tzapu, 2016-07-08:
hi, this is a very good idea, it s not implemented yet, i will consider adding it
although, there might be some other changes when i get around to sorting them out that might make this obsolete.

there s a branch here by @kentaylor that might give you what you want 
cheers

kentaylor, 2016-07-08:
While collecting other settings is a much discussed and popular feature I don't think it ought to be done with WiFi Manager. 

WiFi Manager is useful because until a device connects to a network there is no way to talk to it. Once it gets on to WiFi though everything is different. A better mechanism for configuring the device, once online, is through a web service. Set the credentials on a web site using the browser and the device reads the configuration settings from there.

Not everyone will agree of course.

Humancell, 2016-07-08:
So the *real* question in this issue is to find a way to have the default SSID and Key allowed to be passed into the On Demand Portal so that the user does not HAVE to enter them each time.  For example, I can then always have values that will connect by default to a AP that is packaged with the devices.

Second to this is the ability to add other configuration items, and here I know we disagree.  This is one of the most incredible features of the current WiFi Manager, and we use it extensively.  It allows the use to set security parameters and keys for secure communications with the server, and also other server details.

The issue with your example is that you assume that the DNS name of the server is fixed and coded into software ... and our is not.  The user installing the device can use a default config server, or any other one they want to.

Display of other information is also critical as we can currently display the firmware version, etc.

I"m hoping that the intention is not to remove these features?

closed 2018-01-26 by tablatronix