tzapu/WiFiManager #193: Create Access Point for User don't have modem to connect Internet

tarzan115, 2016-07-08:
when Start ESP8266, WiFiManager Create an Access Point but if I don't have a modem or my modem is offline, I can't connect to ESP8266 throw Access Point to control it. because WiFiManager just create an Access Point to config SSID and Password when I have a modem.
please tell me how to fix that.

tzapu, 2016-07-08:
hi,

i am truly sorry, but i do not understand your issue. could you perhaps describe in more detail what is happening?

what modem?

thanks

tarzan115, 2016-07-09:
I apologize for my bad English.
When I use your library, I can use Autoconnect function to connect to a wifi "AutoConnectAP" or it will create an AP. I mean, if I don't have a Wifi Modem, how can I connect to ESP8266 to control it via the AP that WiFiManager was created. because I see that AP can only access the web to fill in the information of Wifi and connect to my Wifi back. If I turn off my Wifi then I can not directly control ESP8266 via the AP that ESP8266 created.

Here is my code and my video: https://drive.google.com/folderview?id=0B4pX10IagxKia0pOM1dsVl9TbEE&usp=sharing

kdobrev, 2016-07-13:
What tarzan115 is trying to say is my problem too. Imagine I have a protable espp8266 and I need to use it at home and outside. At home there is WIFI hotspot that I connect to, but outside, if there is no accessible hotspot we cannot use the esp8266. WIFIManager does not go into loop(). It just sists and waits to connect it so AP. What we need is one more option that says Create AP in the WiFiManager menu and lets us create AP with dhcp on the esp8266.

tzapu, 2016-07-13:
hi, you can always set a timeout so it quits after keeping the setup open for 1-2 minutes, or alternatively, have the config only come on on demand. 

both docs and examples for either scenario are available in the repo

let me know how you get on
cheers

tarzan115, 2016-07-14:
but when failed to connect and hit timeout in your example "AutoConnectWithTimeout" it call function ESP.reset() so ESP8266 reset and repeat all again :/
maybe you can create in 192.168.4.1 a button for user choose option "Don't have Internet connection" and we use the AP that WiFiManager created.

tzapu, 2016-07-14:
hi,

if you remove the reset, it will just drop back to your sketch as normal

cheers

tarzan115, 2016-07-14:
>

tarzan115, 2016-07-14:
> hi, if you remove the reset, it will just drop back to your sketch as normal cheers

thank you so much :smile: 
I understood :heart: 

thank you for greatest Library. :smiley:

closed 2018-01-26 by tablatronix