tzapu/WiFiManager #194: What happen if using Wifi Manager in restaurant?

vukidrock, 2016-07-12:
Hi there, I love this lib very much, so I have some question:

- Using Wifi Manager lib in restaurant: I have a wifi system before, I use this lib Arduino with ESP8266 like a captive portal to redirect customers to my restaurant website. It will very nice if Wifi manager save my SSID's wifi restaurant and not show other wifi system. 

More specific, when customers turn on wifi on their device, they connect to my restaurant wifi system, they see a website (or Facebook, Twitter... randomly), wait 5 seconds and a button will show to click, after that, they can use wifi system normally.

Anybody think about it? How can I do?

Thank you.

closed 2016-07-13 by vukidrock

probonopd, 2017-08-29:
How did you solve this?

tzapu, 2017-08-29:
hi, isn t it easier just to use a router that has this ability? seems a bit long winded and outside the scope of wifimanager to do this...

tablatronix, 2017-08-29:
Yeah i would just use openwrt running on a cheapo wall wart router

probonopd, 2017-08-29:
After re-reading the original question I agree.