tzapu/WiFiManager #196: EEPROM Storing for SSID/Pass

LorelaiNerv, 2016-07-18:
Hello.
Can you please tell me if there is a special start bit for EEPROM write for the network ssid and pass, or is it 0?
I would like to store some more data in EEPROM and i want to leave that SSID/pass area alone (say.. 50 addresses).
Awsome work with this library! Congrats!

tzapu, 2016-07-18:
hi,

thanks,

you can safely write to eeprom, it is separate from ssid/pwd store

cheers

LorelaiNerv, 2016-07-18:
Great!
Thank you very much,
Keep up the good work ;)

pieman64, 2016-07-18:
@LorelaiNerv https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md shows the memory structure for ESP's and as @tzapu states ssid/pwd are in the WiFi area not the EEPROM area. Do you really want to use EEPROM in the traditional way or would it be better for you to add extra parameters to WiFiManager?

tzapu, 2016-07-18:
@pieman64 thanks

also, there is the additional option of using wifimanager for the parameters, and having it store all in eeprom.

i ve got a sketch somewhere doing that

cheers

LorelaiNerv, 2016-07-18:
I want to use it absolutely old fashioned way.. made some successful tests on arduino mega and now i want to implement the solution to ESP. If i am successful i will post it heere.
My stored values are absolutely independent of this library, bbut i would not interfere with it's functionality.
I will keep you posted.
Again, thank you very much!

pieman64, 2016-07-18:
OK I just worry about some users trashing their devices by posting a bad sketch that exceeds the maximum allowable writes to EEPROM, within a few minutes.

LorelaiNerv, 2016-07-18:
Of course it is a possibility.. but, i have tested the code for some time now.. and looks ok.
My worry was about the address allocation but i saw that it wasn't the case.
Anyway.. average read/writes to eeprom is around 100k if i remember... should take a while.. except for some loop..

LorelaiNerv, 2016-07-18:
On a second thought.. can you please post the example code for storing parameters to wifi area? .. could be a valid solution due to the above mentioned life expectancy of eeprom....
I have "work in progress" here, and i would give it a try.. can save some heads ;)
thank you,

rafaelmaeuer, 2017-06-08:
push! Same for me, please post a working example!

closed 2018-01-26 by tablatronix