tzapu/WiFiManager #197: The SSID and password saved

tarzan115, 2016-07-19:
Thank you for your library!

I have a question, after the Autoconnect function to save the password and SSID. In the loop() function I want to access the SSID and password saved, because I have lost the connection to the WiFi and now I want to check out and reconnect to Wifi in Loop() function.
My question is: How can I get the SSID and password information stored earlier and if I want to connect again, can I use autoconnect function?

thank you :smile: 
and sorry for my bad English.

tarzan115, 2016-07-19:
Sorry, I solved the problem! :smiley:

tarzan115, 2016-07-19:
Hi,
I have checked. WiFiManager does not save SSID and PASSWORD in EEPROM. so where do you saved SSID and PASSWORD? Where can I get them?

Thanks.

skx, 2017-02-01:
You can get them like so:

    Serial.println( WiFi.SSID() );
    Serial.println( WiFi.psk() );

closed 2018-01-26 by tablatronix

rickyanwar, 2021-08-25:
```
Serial.println( WiFi.SSID() );
Serial.println( WiFi.psk() );
```
not work when esp is restart