tzapu/WiFiManager #200: _tryWPS                 = true;

johnpica, 2016-07-23:
If I set _tryWPS                 = true;
and the WPS fails then the configuration of the access point doen't have an IP

*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
***WM: START WPS
*WM: END WPS**
*WM: SET AP STA
Entered config mode
0.0.0.0
ESP13799641
*WM: 
*WM: Configuring access point... 
*WM: ESP13799641
***WM: AP IP address: 
*WM: 0.0.0.0**
*WM: HTTP server started

tzapu, 2016-07-24:
hi

this is exactly why i have not made the function public, have not found a way to make it work if it fails to connect yet

alex

johnpica, 2016-07-24:
Hi
I use this workaround... upload first the sketch example ESP8266WiFi/WifiAccessPoint 
after I upload my sketch... It looks like if I do like this on my esp8266-13 I get IP if WPS fails

 //WiFi.mode(WIFI_STA); //comment
  delay(1000);
  // Called to check if SSID and password has already been stored by previous WPS call.
  // The SSID and password are stored in flash memory and will survive a full power cycle.
  // Calling ("",""), i.e. with blank string parameters, appears to use these stored values.
  WiFi.begin("","");
  // Long delay required especially soon after power on.
  delay(7000);
  
  // Check if WiFi is already connected and if not, begin the WPS process. 
  if (WiFi.status() != WL_CONNECTED) {
      Serial.println("\nAttempting wifi connection ...");
      WiFi.beginWPSConfig();
      // Another long delay required.
      delay(3000);
      if (WiFi.status() == WL_CONNECTED) {
        Serial.println("Connected!");
        Serial.println(WiFi.localIP());
        Serial.println(WiFi.SSID());
        Serial.println(WiFi.macAddress());
      }
      else {
        Serial.println("Connection failed!");
        //creem un captive AP
        Serial.print("Configuring access point...");
        //WiFiManager
        //Local intialization. Once its business is done, there is no need to keep it around
        WiFiManager wifiManager;
        //reset settings - for testing
        //wifiManager.resetSettings();
        //set callback that gets called when connecting to previous WiFi fails, and enters Access Point mode
        wifiManager.setAPCallback(configModeCallback);
        //fetches ssid and pass and tries to connect
        //if it does not connect it starts an access point with the specified name
        //here  "AutoConnectAP"
        //and goes into a blocking loop awaiting configuration
        if(!wifiManager.autoConnect()) {
          Serial.println("failed to connect and hit timeout");
          //reset and try again, or maybe put it to deep sleep
          ESP.reset();
          delay(1000);
        }

      }
  }  
  else {
    Serial.println("\nConnection already established.");
  }

johnpica, 2016-07-24:
ignore my last comment... the problem now is that the WPS function is not working anymore.

malebuffy, 2017-04-29:
found a workaround..
make a fake connection before trying to reconnect with wifimanager after a failed wps connection

`WiFi.begin("foobar",""); // make a failed connection`

closed 2018-01-26 by tablatronix