tzapu/WiFiManager #201: Sending station mode ip

cotuex, 2016-07-24:
Hello, I'm building a robot using the ESP8266 and arduino, but it will not use the computer, it will be a standalone version. My problem is that when I finish connecting the ESP to my network I don't know the ip of my ESP. Can someone make a function or an adaptation for that or simply tell me how can I do what I'm looking for? Thank you.

tzapu, 2016-07-24:
hi,

you can make it do a http request at the end to report the ip to some other entity, or display it on a display if you ve got attached.

where exactly would you want the ip displayed?

cheers

cotuex, 2016-07-24:
I would like to make this robot without needing a wifi network, but with the posibility to connect to one. For that reason, it will create a TCP server so you can send commands to it like speed, direction, etc. My problem is that when you can succesfully connect, if the ESP is not connected to a computer you don't know which is the ip of the server the ESP has created. What I was expecting is the posibility that using your library, you can enter to the wifimanage from my cellphone, to connect it to my local network, and that finally, when I press "Connect", after the ESP does all what it has to do, to redirect me to a page telling me which will be my ESP wifi ip. If you don't understand something, ask me.

tzapu, 2016-07-26:
hi, unfortunatelly, the way wifi manger and the esp8266 itself works, that will be impossible to achieve in most cases. mostly because if the AP you are connecting the ESP to and the AP the ESP creates are not on the same channel, your phone will get a disconnect from the ESP AP, hence no page refresh is possible.

pieman64, 2016-07-26:
@cotuex how are you going to control the robot? Presumably it will be with WiFi via the ESP8266. If this is the case then you will be using WiFi and the IP address can be made available to you in many ways. With [Blynk](http://www.blynk.cc/) it can be sent to your Smartphone via a Push message, email, SMS or Twitter message etc. Perhaps if you can explain in more detail how you are going to control the robot we can offer a possible solution.

kentaylor, 2016-07-27:
The version of WiFi manager at  https://github.com/kentaylor/WiFiManager/ will show the IP address on the default page as [shown]( https://drive.google.com/file/d/0B3RMBzgiNpwGd01FOXo3MkN6ck0/view?usp=sharing). This version of WiFi manager keeps the web server running until it is explicitly told to close. The IP address also available in json format at http://192.168.4.1/state .

As stated by @tzapu the WiFi connection will drop if the the network to which it connects is on a different channel. It will then restart on the new channel. The browsing device may swap to an alternative WiFi network rather than the ESP8266 and if the web pages are being accessed through a device portal configuration screen then the screen will close. If it is accessed through a normal web browser it is more intuitive as the page will not disappear. 

If it has switched WiFi networks you will need to reconnect to the ESP network to access the web server again. The web server will also be visible at the device IP address on the network to which it has connected. 

This can be confusing but it is difficult to make it intuitive as the browsing device can not be stopped from jumping off the ESP8266 WiFi network. In my view the @tzapu version is confusing as well because you get no feedback on whether the ESP was able to connect. If it didn't, most likely because the password was wrong, you never know why and many people interpret the sudden and unexpected closure of a portal as a failure. They try to reconnect to the network but it is no longer there even though many browsing devices will report the network as still existing.

An app would be a good way to hide the complexity from users while still allowing configuration from a browser for those that understand the nuances.

cotuex, 2016-07-27:
@kentaylor, that's what i'm looking for, some feedback. The designs of my robot are unnecessary to know, but if you want I can paste the code later (I'm not using my computer right now). What i'm looking for is some feedback, like the ip if it can sucessfully switch to a wifi connection. I know a bit of c# to use Xamarin so I'll use a tcp server on the esp8266 and i'll make an android application to connect the esp to wifi and then wait for feedback. If I make some progress I'll write here soon.

cotuex, 2016-07-27:
@pieman64, I won't use blynk because I'm making this robot independent of wifi connections. If you can connect it to a wifi network, no problem, but if you can't, my idea is that the esp can make a softap network with a tcp and you can control the esp using connecting to that network without using wifi.

pieman64, 2016-07-27:
@cotuex if you are a competent Android application programmer then you might not need Blynk but just so you are aware:

1. It can use Bluetooth
2. The server can be on your robot e.g running on a $5 Raspberry Pi.

Still confused how you are planning to control the robot without a some form of network. 
Good luck with your project.

tzapu, 2016-07-27:
i think he will have a connection between esp and phone, and that s it, no real AP involved.

i believe blynk have a direct connection feature in the works, not sure if it s done or not, with which you wouldn t need a server in between. 

thanks guys for all the info. maybe ken s branch will work for his purpose.

cotuex, 2016-07-27:
@pieman, I will try to explain it more clearly. I want to control my robot (a kind of rc car) via wifi using the esp. For this purpose, I thought TCP sockets would be a good, direct, not laggy way to control it. I want to, some way, connect to the esp and decide if I'll connect it to my wifi network, or it will start the TCP server in softap mode, so I can control it in my school, for example, where I don't have wifi. To control my robot, I'm going to use my cellphone. I'm going to create a program for android so I can send tcp sockets to the esp, wich will be sent to the arduino that is going to control the motors, the direction, etc. In the afternooon I'll upload my code and the schematics of my project.

hmiezin, 2016-11-03:
I think I had a more or less similar question (was not a robot, but a remote temp sensors that i wanted to configure easily from any phone or tablet or computer).
I choosed to trig a config mode on the with with button (keeping a button pressed when powering).
In the config mode, the iot goes to Station Mode only, with a fix SSID, and generate html pages with forms, which allows an easy configuration. The configuration is then stored in a SPIFFS text file by the iot and remains available for the real application. 
Of course, this is not limited to configuration ans SPIFFS files only one option.
I found this very convenient, especially to define SSID ans PSK when an iot joins a new network..
If it covers your needs, do not hesitate to ask, I can provide sample code.

kentaylor, 2016-11-03:
@hmiezin, rather than offering to supply code, better to just stick it on github and point to it. 

There is [an example](https://github.com/kentaylor/WiFiManager/tree/master/examples/ConfigOnSwitch) that will create a configuration portal on a button press at any time. Holding down a button at startup would make a good example to add there as the code in the sketch would be simpler. If you do one I'll add it to the examples. Hava a look at @datacute's double press reset branch to see how.

[IOT configurator](http://configure.urremote.com/) is another way to configure your device once a WiFi connection has been established.

closed 2018-01-26 by tablatronix