tzapu/WiFiManager #207: Error when compiling AutoConnectWithFSParameters

Daemach, 2016-08-02:
1.6.7 IDE
2.3.0 ESP Arduino
Current WiFi Manager code

I'm getting the following errors when trying to upload the AutoConnectWithFSParameters sketch:

WARNING: Spurious .github folder in 'ArduinoJson' library
C:\Users\johnw\Dropbox\Private\Arduino\Sketches\libraries\WiFiManager\examples\AutoConnectWithFSParameters\AutoConnectWithFSParameters.ino: In function 'void setup()':

AutoConnectWithFSParameters:76: error: 'WiFiManagerParameter' was not declared in this scope

   WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);

   ^

AutoConnectWithFSParameters:76: error: expected ';' before 'custom_mqtt_server'

   WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);

                        ^

AutoConnectWithFSParameters:77: error: expected ';' before 'custom_mqtt_port'

   WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 6);

                        ^

AutoConnectWithFSParameters:78: error: expected ';' before 'custom_blynk_token'

   WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 32);

                        ^

AutoConnectWithFSParameters:85: error: 'class WiFiManager' has no member named 'setSaveConfigCallback'

   wifiManager.setSaveConfigCallback(saveConfigCallback);

               ^

AutoConnectWithFSParameters:88: error: 'class WiFiManager' has no member named 'setSTAStaticIPConfig'

   wifiManager.setSTAStaticIPConfig(IPAddress(10,0,1,99), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

               ^

AutoConnectWithFSParameters:91: error: 'class WiFiManager' has no member named 'addParameter'

   wifiManager.addParameter(&custom_mqtt_server);

               ^

AutoConnectWithFSParameters:91: error: 'custom_mqtt_server' was not declared in this scope

   wifiManager.addParameter(&custom_mqtt_server);

                             ^

AutoConnectWithFSParameters:92: error: 'class WiFiManager' has no member named 'addParameter'

   wifiManager.addParameter(&custom_mqtt_port);

               ^

AutoConnectWithFSParameters:92: error: 'custom_mqtt_port' was not declared in this scope

   wifiManager.addParameter(&custom_mqtt_port);

                             ^

AutoConnectWithFSParameters:93: error: 'class WiFiManager' has no member named 'addParameter'

   wifiManager.addParameter(&custom_blynk_token);

               ^

AutoConnectWithFSParameters:93: error: 'custom_blynk_token' was not declared in this scope

   wifiManager.addParameter(&custom_blynk_token);

                             ^

exit status 1
'WiFiManagerParameter' was not declared in this scope

Daemach, 2016-08-03:
This problem was resolved by using the 2.2.0 esp arduino core rather than 2.3.0.  The web page is still empty when lpaded.

tzapu, 2016-08-04:
hi, i am compiling with 2.3.0 and even with the latest github, just fine. it seems to not see the parameter class, which is weird, as it s in the same file...

can you check you don t have another version of the library except the latest somewhere?

regarding blank pages, i ve usually had that when not setting the flash size in arduino ide to the proper flash size of the esp module

cheers

Daemach, 2016-08-04:
No extra functions.

The web pages are the biggest problem. I'm set to 4M (3M SPIFFS) and have never had a problem until this library.

closed 2018-01-26 by tablatronix