tzapu/WiFiManager #208: Cannot set hostname or make mDNS work

kdobrev, 2016-08-04:
My esp8266 seems to always have hostname ESP_something as a name.

My setup is:
`void setup()
{
  WiFiManager wifiManager;  // commented here because it is initiallized outside setup()
  wifiManager.setAPCallback(configModeCallback);
  if (!wifiManager.autoConnect("KDO")) {
    Serial.println("failed to connect and hit timeout");
    ESP.reset();
    delay(1000);
    Serial.println("connected...yeey :)");
  }

  WiFi.hostname(host);
  wifi_station_set_hostname("KDO");
  if (!MDNS.begin(host)) {
    Serial.println("Error setting up MDNS responder!");
  }
  server.begin();
  MDNS.addService("http", "tcp", 80);`

Here I use 3 thing that according to me should set my hostname, but it is not set and my device is not reachable by hostname on the local network.
in the library itself I cannot find wifi_station_set_hostname
Can I put is somewhere?

gordonthree, 2016-08-13:
Hostname registration seems hit and miss. A lot of consumer wifi routers use DNSMASQ for dhcp and local dns caching, and it seems picky about registering certain hostnames and ignoring others. I haven't had a chance to dig into the dnsmasq docs, to understand how hostname registration works.

I have the same code running on ten different ESP, exactly the same model. Some of them register as their hostname, some register as ESP_xxxxx and some don't register any hostname at all (show up as * in dnsmasq).  I can't figure out why this happens, since it's identical code running on identical model ESP.

The variable "host" in your example, is it char, const char or String?

I've tried invoking WiFI.hostname() before and after WiFiManager, it didn't seem to make a difference.

I guess this puts me in the "me too" category.

kdobrev, 2016-08-18:
At work I think there is no chance to get a hostname in my corporate network, but at home I get these ESP_xxxxx. What I found out in a threat here was that we need to place wifi_station_set_hostname("KDO"); before any network operation has occurred so my code currenlty is:
```
const char* host = "CloudLamp";
//other stuff
void setup()
{
  Serial.begin(115200);
  //pin stuff
  //display stuff
  wifi_station_set_hostname("CloudLamp");
  WiFiManager wifiManager;  // commented here because it is initiallized outside setup()
  wifiManager.setAPCallback(configModeCallback);
 if (!wifiManager.autoConnect("CloudLamp")) {
    Serial.println("failed to connect and hit timeout");
    ESP.reset();
    delay(1000);
    Serial.println("connected...yeey :)");
  }
WiFi.hostname(host);
  if (!MDNS.begin(host)) {
    Serial.println("Error setting up MDNS responder!");
  }
  server.begin();
  MDNS.addService("http", "tcp", 80);
 //and so on
```
I still haven't tested this at home to check if it works.

@gordonthree You say it is up to the router to get a hostname or not from our ESPs and some routers get it and some don't? Or these ESPs you talk about are in one and the same network?

closed 2018-01-26 by tablatronix