tzapu/WiFiManager #211: Redirecting after connection

Daemach, 2016-08-07:
I want users to be able to tie their devices to user accounts on an external webserver so they can see their own data streams.  After wifi is configured, what would it take to generate a redirect to an external web site that included a dynamic parameter, allowing 15-20 seconds for the phone to reconnect to wifi/3G after the ESP access point goes offline?  The dynamic parameter I need at the moment is the MAC concatenated to the chipID.  (https://my.website.com/register.aspx?sn=MACADX+CHIPID)  I'm using that string as a device serial number.  If the ESP can pass that along, the user should just need to log into the site and confirm.

Daemach, 2016-08-07:
I've tried using the custom header method.  The function doesn't like strings, though:/

```
String tempURL = "<script>window.setTimeout(function(){ window.location = \"" + redirectURL + SN + "\"; },3000);</script>";
  wifiManager.setCustomHeadElement(tempURL);
```

kentaylor, 2016-08-09:
I don't think you can reliably redirect to a different WiFi network.

Rather than redirect after the WiFi Manager page, use a web browser to access WiFi Manager at  http://my.website.com/ . Then you will see WiFiManager when on the ESP8266 network and http://my.website.com/  when there is an internet connection. Note that it must be http not https as https can not be redirected.

As suggested in https://github.com/tzapu/WiFiManager/issues/212#issuecomment-238466227  (http://my.website.com/register.aspx?sn=MACADX+CHIPID)  is where you should read your blynk token from.

closed 2018-01-26 by tablatronix