tzapu/WiFiManager #212: Can't paste into captive portal fields with iPhone

Daemach, 2016-08-08:
On my older android, I don't get a captive portal page immediately when connecting to wifi.  When I open the browser, I do get redirected to the config page as expected.  On my iphone 6, clicking on the AP in the wifi setup screen opens the captive portal page immediately, but I can't paste anything into the fields on this screen.  I had hoped to paste the blynk token from the email they send, and it looks like that's what you intended.

For simple wifi config, the captive portal is great.  For folks that need to gather more info, it would be great if we could find a way around the current behavior, but still make the config page accessible via the browser.  It's an extra step, but that token is too important to trust to manual entry.

https://discussions.apple.com/thread/7491051?start=0&tstart=0

Can this be done through the DNS server?  If something tries to get to http://captive.apple.com/hotspot-detect.html, the device needs to return 
<HTML><HEAD><TITLE>Success</TITLE></HEAD><BODY>Success</BODY></HTML>

kentaylor, 2016-08-09:
The captive portal is just a stripped down web browser as stated on your link https://discussions.apple.com/thread/7491051?start=0&tstart=0. On Android and and Chromebooks this browser does not open automatically but must be clicked. 

You can close this pop up browser and use the regular web browser as an alternative. Then cut and paste will work fine. There is a version of WiFi manager that shuts down the ESP8266 web server manually rather than automatically at https://github.com/kentaylor/WiFiManager/ which I think is more intuitive with a normal web browser.

In my view you should get your blynk token from a server somewhere and only use WiFi manager for connecting to WiFi.

Daemach, 2016-08-09:
I need some way of linking the hardware device to a user's account.  This seems to be the simplest place to make that link, either via a stored token or a redirect with a serial number.

mfalkvidd, 2018-01-26:
@tablatronix is this fixed now? If so, do you have a commit reference?

tablatronix, 2018-01-26:
Needs confirmation

tablatronix, 2018-01-26:
cannot reproduce