tzapu/WiFiManager #216: AP mode stuck on after STA re-connected

mtnbrit, 2016-08-13:
Hi, Im seeing my devices are all showing their config APs after they lost connection to their normal AP today. So they lost connection to the stored AP in STA mode, after the timeout, the config AP is started, then the normal AP comes back and they re-connect in STA mode, but in fact its in AP+STA mode as I still see the APs broadcasting in my wifi networks. 

Even if they are in config mode, clearly if they are able to reconnect to the stored ssid, then config AP mode should be canceled and just go back to regular STA mode, no?

tzapu, 2016-08-15:
hi,

that s what should happen, when connected it goes back to STA mode.

do you still see the ap’s after some time? some devices just seem to cache the list of avaiable APs for a while…

cheers

mtnbrit, 2016-08-15:
Definitely it stays in AP+STA mode, still see the ssid permanently.

kentaylor, 2016-09-25:
Your issue reads to me as:-
1. AP is missing so as specified by autoconnect go into config mode.
2. AP comes back, connects to network but device stays in config mode.

If so, that is how WiFiManager is programmed to behave. It will connect back to the network because it is simultaneously in AP and STA mode as configured at  https://github.com/tzapu/WiFiManager/blob/028bdeeb17ef12b8dc0c2fc05fc3ae4bc102ba70/WiFiManager.cpp#L154 but remains in config mode. That is an undesirable situation in my view. You can put a timeout on config mode so it will drop out of config mode after a while which may work for you.

You would like it to go out of AP mode when it successfully reconnected and it could be programmed that way but there is problems with that logic. It would then become impossible to change networks when a network to which it can connect is visible and many applications should still function even without a WiFi network so for those applications you don't want it to jump into config mode every time the WiFi network goes away. 

I don't think there is any way to automatically know when it should go into config mode, only a human has the knowledge of whether a new network is required or it is better to wait for the network to come back. Therefore you require a human to initiate config mode and here is an example that works that way  - https://github.com/kentaylor/WiFiManager/tree/master/examples/ConfigOnSwitch .

closed 2018-01-26 by tablatronix