tzapu/WiFiManager #217: Custom SSID for Captive Portal in WiFi selection

kfeger, 2016-08-19:
Greetings!
First of all: Great, great work!!!!
One question: How can I asign a custom ssid for the captive Portal that's set up when doing an autoconnect as you show it e.g. in AutoconnectWithFeedback.ino (for display reasons I need the callback feature)?
What I try to get is an ssid that is not the default "ESP[chipID]" but rather "MyConfigWiFi" or the like.
Thanks and greetings from Dresden, Germany!
Karl

tzapu, 2016-08-20:
hi,

just send the SSID as the first parameter to autoConnect

thanks und danke

kfeger, 2016-08-20:
Thanks!

I misinterpreted the time, the debug output was generated. That was to easy…

Thanks again!!

Karl

skx, 2017-02-01:
@kfeger - Perhaps time to close this issue?

kfeger, 2017-02-05:
You’re definitly right. I’ll do it right away.

BTW: Works like a charme!!!

Karl

closed 2017-02-05 by kfeger