tzapu/WiFiManager #218: Configuration by app

lunanigra, 2016-08-22:
Hello, your WiFiManage is quite cool :-)

But the configuration via Browser is not really comfortable. The user has to know several things (connect to ESP8266, open browser with right URL and so on). Then the ESP8266 is often quite slow...

A colleague has implemented a similar (but slightly different) approach. An Ionic app (can be native as well) connects to the ESP8266 if not already configured. This approach reduces the communication and allows developing a complex (or more comfortable) UI.

Is this something that sounds interesting? If so I can send you a copy of our Ionic app... Then you can check how to merge with your solution.

Alkia, 2016-08-23:
Looks a great idea.

Kind regards,

tzapu, 2016-08-23:
HI,

thanks, it is a great idea, unfortunately it can t sort out some of the biggest issues with the whole process. (or at least i don t know how). You would still need to connect to the esp manually. On most devices(or the ones i have tested), after that step, the configuration portal opens automatically, so you shouldn t need to know IPs anymore. DNS redirection should also work to direct any address in your browser to the proper IP.

In my case, i am using an iphone to configure my devices, i connect to the ESP network, captive portal pops up, enter details, save and that s it... and the only annoying step is that i have to connect to the wifi network, but that s unavoidable currently. 

Now, if any of the smart config options or the WPS would be reliable and work, it would be quite a different thing altogether, but, again, in my experience, they are quite hit and miss...

I would love to see the app, maybe there s some scope to link the two together, make WiFiManger also work with an app 

thanks again

lunanigra, 2016-08-25:
Hi, can you give me an email address? I will send you our sample code.

Beside a faster UI the solution via app will not require manually changing WiFi setting. Similar to Google Cast the app will scan for ESPxxxx networks. If available a setup screen will be provided. Only WiFi settings will be transferred; no HTML or other UI elements. This will speed up configuration and also safe resources on ESP.

Would be great if we can link both approaches together :-)

Thanks, JC

tzapu, 2016-08-25:
HI,

On iOS at least, you can t send the settings without connecting to the actual wifi. You also can t force a connection and I am not sure even if you can scan for wifi networks, last i checked there were no public apis for the WIFi connection management in iOS.

Google cast employs a number of other technologies that are most likely not available on the ESP, and even those don t work reliably for me at all times. WifiDirect, Multipeer, quite possibly even bluetooth.

You can use: alex@tzapu.com

Cheers
Alex

tzapu, 2016-08-25:
> You can use: alex@tzapu.com <mailto:alex@tzapu.com>

Forgot, i think chromecast is even putting out or listening for an out of audible range audio signal through the tv and links to the phone through that to send settings, etc.

lunanigra, 2016-08-25:
hmmm...

<alex@tzapu.com>
    (reason: 552-5.7.0 This message was blocked because its content presents a potential)

tzapu, 2016-08-25:
seems it doesn t like your attachment ..

lunanigra, 2016-08-25:
Seems so... But it's just a ZIP ;-)

umranium, 2016-08-28:
Hi,

We have an Android app already. We're currently testing it and fixing some minor issues. If you'd like to test it out, please take a look: [ESP Connect on Google Play](https://play.google.com/store/apps/details?id=au.com.umranium.espconnect)

tzapu, 2016-08-29:
oh man, that s so cool.
i would like to make a batter API for this, so anyone can have/build or use an app and also have a showcase/apps section on the main github page.
is there anything in particular you d like supported in the api? Would you be ok on being featured?
i love it when things develop further :)

Alkia, 2016-08-29:
Looks great,

Looking forward helping for test.

It would be great if the ionic app could natively requests for 2 optional
feilds:
- an email address
- a Pin code

Kind regards,

Maxime

umranium, 2016-08-29:
@tzapu We added some APIs and some fixes to the code to get it to work smoothly with apps. For example, supporting a more machine-centric data format (we went with JSON). You can find the additions (and fixes) in this repository: [Ken Taylor's WiFiManager repo](https://github.com/kentaylor/WiFiManager)

umranium, 2016-08-29:
@tzapu Forgot to answer one of your questions. Yes, we would **love** to be featured! :)

lunanigra, 2016-08-29:
Sounds cool!! Curious about testing the next release :-)

----- Nachricht beantworten -----

pieman64, 2016-08-29:
@umranium can we have a Blynk token field?

tzapu, 2016-08-30:
i think ideally i would expose some api endpoint, and all apps should fetch all configured fields in the sketch and allow configuration

cheers

umranium, 2016-08-30:
@pieman64 We might look into that in the future. Currently our scope is limited to getting the user's device connected to the WiFi (and therefore the internet) as painlessly and as effortlessly as possible.

umranium, 2016-08-30:
@lunanigra I should have an update out by early next week.

TechieIndianGuy, 2016-09-22:
@umranium Has the update released?

umranium, 2016-09-25:
@TechieIndianGuy I just updated the alpha testing version. Here's the link to opt into the alpha testing:

[ESP Connect Alpha Testing Opt In Link](https://play.google.com/apps/testing/au.com.umranium.espconnect)

I'm keen to get any feedback.

Melcos1970, 2017-05-08:
I have made a small change to WiFiManager.cpp and .h to get a mid-term solution to the problem with WPS.
It's not the best way, but now you can use WPS if you are going to ..
I have no idea how to add suggestions to github, but I want to post them here so can add them if you want to use them.
The solution is actually simple .. I just add a button extra that makes a WPS call.

Changed:
Const char HTTP_PORTAL_OPTIONS [] PROGMEM = "<form action = \" / wifi \ "method = \" get \ "> <button> Configure WiFi </ button> </ form> <br/> <form action = \" / 0wifi \ "method = \" get \ "> <button> Configure WiFi (No Scan) </ button> </ form> <br/> <form action = \" / i \ "method = \" get \ "> <Button> Info </ button> </ form> <br/> <form action = \ "/ r \" method = \ "post \"> <button> Reset </ button> </ form> <br/> <Form action = \ "/ w \" method = \ "post \"> <button> WPS Setup </ button> </ form> ";

Inserted: lines +++
    Void handleReset ();
+++ void handleWPS ();
    Void handleNotFound ();

  Server-> on ("/ r", std :: bind (& WiFiManager :: handleReset, this));
+++ server-> on ("/ w", std :: bind (& WiFiManager :: handleWPS, this));
  // server-> on ("/ generate_204", std :: bind (& WiFiManager :: handle204, this)); // Android / Chrome OS captive portal check.
  Server-> on ("/ fwlink", std :: bind (& WiFiManager: handleRoot, this)); // Microsoft captive portal. Maybe not needed. Might be handled by notFound trades.

Inserted section / ++ ++ / after void WiFiManager :: handleReset () section
/ ++
Void WiFiManager :: handleWPS () {
  DEBUG_WM (F ("WPS Start"));

  String page = FPSTR (HTTP_HEAD);
  Page.replace ("{v}", "Info");
  Page + = FPSTR (HTTP_SCRIPT);
  Page + = FPSTR (HTTP_STYLE);
  Page + = _customHeadElement;
  Page + = FPSTR (HTTP_HEAD_END);
  Page + = F ("Module will start WPS in a few seconds.");
  Page + = FPSTR (HTTP_END);
  Server-> send (200, "text / html", page);

  DEBUG_WM (F ("Sent WPS page"));
  // WPS works only in STA (Station mode) only.
  WiFi.mode (WIFI_STA);
  delay (1000);
  startWPS ();

[WiFiManager_WPS_Files.zip](https://github.com/tzapu/WiFiManager/files/983659/WiFiManager_WPS_Files.zip)

  delay (5000);
  ESP.reset ();
  delay (2000);
}
++ /

FernandoGarcia, 2017-07-18:
Hi!

@umranium why your App doesn't connect to ESP when there a SSID and Password configured here?

`if (!wifiManager.startConfigPortal("ESP8266","password"))`

It's always saying "connecting to ESP....."

But it works in this way:

`if (!wifiManager.startConfigPortal())`

I would like to post this issue to your repo but it's not available.

Best regards.

erdemontas, 2017-08-09:
I have working device and I'm using wifimanager library properly. I also have an android app, I want to implement"configuration by app" rather than having @umranium 's app. What kind of changes should I do?

pieman64, 2017-08-09:
@erdemontas you might want to look up **Blynk provisioning** but it's difficult to tell what you want to do from your post.

erdemontas, 2017-08-10:
@pieman64 let me try to put it this way; what I want to do to connect my esp to internet via my app not using web portal. 
First idea which came to my mind is, to be successful on this task, I have to deliver the credentials to my esp and save them in ESP. I figured out how to save them but where I'm stucked is how should I deliver them from Android app to Esp(I guess it has to be in AP mode)

kapiloberoi, 2017-08-12:
I have been able to put together an app that autodetects the esp8266 and let you select your wifi network settings to be saved into the esp. The only think now remains is to set a hostname so that ince config is over we can access it by name instead for looking for the ip address.

Used appinventor2 and credit goes to Taifun for excellent wifi extension. "Kapiloberoi@gmail" and ill send the code along with apk. Will also post the code and app here in a day !! Be warned that im a terrible coder:))

Cheers -

gopchi, 2017-11-08:
Guys, I am not able to get the app to find my ESP... What am I doing wrong? Installed the WiFi Manager library then used the AutoConnect example. The ESP creates an AP but when I scan with the app @umranium provided says Unable to find any ESP8266 nodes. 

@FernandoGarcia  How did you set up your ESP?

Would appreciate some help, because I really want to try and configure my esp's through the mobile app. :)

anshulfd, 2017-12-18:
@tzapu Are there any plans to add the features from https://github.com/kentaylor/WiFiManager to this repo?

tablatronix, 2017-12-18:
Features yes, merge no.

closed 2018-01-26 by tablatronix

BuzzHari, 2018-06-25:
Is an open source app for configuring esp8266 available yet?

DeveloperChand, 2018-10-16:
@gopchi 
facing same issue. any solution available

laurensV, 2023-01-04:
Would also like to know if an open source version of an app is available (auto detect esp etc.)