tzapu/WiFiManager #223: Captive portal Question

kiralikbeyin, 2016-09-03:
My network don't need a password, how to stay in the network when i close captive portal?

i have a welcome page and a index.htm page. Captive portal serves welcome page.

When i quit from portal i lost connection because i never connected?

+ index.htm's javascript is not working. (welcome page has a index.htm link)

kentaylor, 2016-09-25:
I think you are are saying the network you are trying to connect to requires you to configure something in a captive portal page on that network. 

WiFiManager doesn't do that.

kiralikbeyin, 2016-09-25:
I was using iphone, it doesn't stay on network if you have a captive portal (android stays and says no internet connection) . For iPhone there is an option when click info "auto connect" for network with a captive portal and use it "without internet connection"
Thanks.

closed 2016-09-25 by kiralikbeyin