tzapu/WiFiManager #226: Improve the UI

akshitgandhi, 2016-09-07:
Hey! Nice job with the library. I wanted to add more style elements on the configuration page, but being new to this I cannot figure how to add more css and JavaScript. I wondered if I could use the template in the extras folder and use the snippets into the script. Please tell me how can I add custom css and JavaScript. I refered the documentation but couldnot understand , please help me!

tzapu, 2016-09-08:
hi,

if you can hold a few more days, i was just working on a nodejs app that exports consts from the the template, so anyone can make it look however they like easier.

just need to make sure it s outputing the proper stuff, then i ll include it

cheers

akshitgandhi, 2016-09-08:
Thanks!
Approximately how much time will it take? Can you share the developmental version so that even I can help you up in bug fixes?

tzapu, 2016-09-08:
hi,

just committed it. check the extras folder, you just run it with node parse (need to have nodejs installed of course)

the next step was to check of the exported consts are valid, reintegrate whatever changes have been made to the consts in the library into the template, exporting, seeing if they are equivalent and then finally modifying wifimanger to use them.

cheers

akshitgandhi, 2016-09-09:
thank you keep up the good work!

closed 2016-09-09 by akshitgandhi