tzapu/WiFiManager #227: Auto connect / Hotspot DHCP problems

ejbp, 2016-09-13:
When the esp8266 doesn't find a wifi to connect, auto connect provides a hotspot but i need to try (sometimes more than 5 minutes after plug/unplug the chip many times) to connect to the hotspot. It seems the problem is that, for some reason, it doesn't provide an ip address to the client. If I try with a static ip in the client side, it works, but this is not production friendly. 

Here's the code:
`
WiFiManager wifiManager;
wifiManager.setDebugOutput(true);
wifiManager.setSaveConfigCallback(saveConfigCallback);
wifiManager.setAPCallback(configModeCallback);
wifiManager.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));
wifiManager.setTimeout(360);
if(!wifiManager.autoConnect("Wifi-box")) {
    delay(3000);
    ESP.reset();
    delay(5000);
  } 
`

tzapu, 2016-09-13:
hi

could you try on another module as well? my connection is usually instant, from an ios device...

also, make sure you set the right flash size for your esp module

cheers

ejbp, 2016-09-14:
Hi, You mean using another module other than WifiManager? Meanwhile, I'll do some debug.
As a note: It happens with any device, android, iPhone, laptop, tablet...

tzapu, 2016-09-14:
hi

i meant another esp8266 module

cheers
alex

ejbp, 2016-09-15:
Hi

After some debug I think I discovered what is causing the problem. 

If you setup/connect to an wifi network with success through your esp8266 that configuration is saved, as expected. Next time you turn on the esp8266, it connects to that same wifi, if available, as expected too. 

The problem starts when esp8266 doesn't find that wifi available anymore, so it starts in the process of trying to reconnect, scan all networks, trying to reconnect and at same time (1s in 1s), trying to reply to serve the hotspot (Station) with all the dhcp, dns, http(captive) services trying to run and reply. If you have a few wifi networks around this is not a problem, but if you have to many, the esp probably doesn't have time enough to process all of those and therefore, doesn't allow you to connect to it, or if it does (with luck) you will have the same problems trying to have have the http captive server answering to you.

I'll try to figure out how to change the 1s to 5s or even 10s and comeback to double confirm that that was the problem.

tzapu, 2016-09-19:
if you think this is the issue, you can add a WiFi.disconnect() before starting the configuration portal to test.

that should stop it trying to connect while in setup mode.

cheers

kentaylor, 2016-09-25:
I doubt the scenario suggested in https://github.com/tzapu/WiFiManager/issues/227#issuecomment-247241799 is occurring because the ESP8266 is put into access point (AP) mode when the portal is set up (see https://github.com/tzapu/WiFiManager/blob/master/WiFiManager.cpp#L154) . In AP mode it doesn't try to connect to another WiFi network so it is not "in the process of trying to reconnect".

In dual mode however it does try to connect to an access point and the WiFi Manager version at https://github.com/kentaylor/WiFiManager/ does use dual mode sometimes. This tells me the the explanation of why it has to switch out of dual mode at https://github.com/kentaylor/WiFiManager/blob/master/WiFiManager.cpp#L204 is wrong and needs to be updated with this explanation.

klemic, 2016-09-25:
Hi,
when using the ESP8266 version 2.3.0 (and ver 2.2.0 too)I have the same issue, the ESP is not connecting to the Wifi, when downgrading to version 2.1.0 it works always...

kentaylor, 2016-09-25:
@klemic there are lots of things fixed in later ESP8266 library versions so I wouldn't be downgrading to get better functionality from WiFiManager. 

Change this line https://github.com/tzapu/WiFiManager/blob/master/WiFiManager.cpp#L154 to WiFi.mode(WIFI_AP); and it will no longer be trying to connect to another network when in configuration mode. That will fix it if the scenario in https://github.com/tzapu/WiFiManager/issues/227#issuecomment-247241799 outlined by @ejbp is correct.

You could try the version at https://github.com/kentaylor/WiFiManager/ which doesn't use dual mode when the configured WiFi network is not available. 

You can also see what has changed between ESP8266 library versions and investigate why older versions work better in this case but that requires more effort than I think it is worth.

klemic, 2016-09-26:
Hi Ken, thanks a lot. However, the problem is not the configuration mode. When using the old library the ESP is just connecting to the last known Wifi and not entering the configuration mode at all. When using the new library the ESP is almost never connecting to the exisiting WIfi and then entering the configuration mode. So there is already something wrong before. 

You're right, the effort to find the reason is a problem, that's why I helped me with downgrading...

kentaylor, 2016-09-26:
@klemic it sounds like your issue is different to the issue described by @ejbp.

The WiFi Manager version at https://github.com/kentaylor/WiFiManager/  doesn't use dual mode when the configured WiFi network is not available so should be a solution to the issue described by @ejbp . 

Also  the fix provided by @lawrence-jeff and discussed at https://github.com/tzapu/WiFiManager/issues/215#issuecomment-249485865 should work also.

klemic, 2016-09-26:
@kentaylor yes, it seems so...

closed 2018-01-26 by tablatronix