tzapu/WiFiManager #229: After Config Portal Timeout ESP always act as Access Point

mocheffendi, 2016-09-29:
Dear Tzapu,

I Use Configuration Portal timeout
wifiManager.setConfigPortalTimeout(180);

But after Timeout occurs the ESP8266 still act as Access Point
How to shutdown the Access Point after Config Portal Timeout.

Thanks
Best Regards

M. Effendi

krzbor, 2016-10-09:
I have a similar problem. When timeout ESP properly restarts, it is running my program, but the AP is still visible. I can connect to the AP. I can not, however, connect with the web: 192.168.4.1.

kentaylor, 2016-10-10:
That is occurring when the esp8266 starts up in dual mode. You can set it to station mode in your code or there is another version of WiFiManager that [Selectively Operates In Dual Mode](https://github.com/kentaylor/WiFiManager#selectively-operates-in-dual-mode).

krzbor, 2016-10-12:
Is it enough to add WiFi.mode (WIFI_STA); before ESP.reset ();?

kentaylor, 2016-10-12:
Better to do it at start up [like this](https://github.com/kentaylor/WiFiManager/blob/master/examples/ConfigOnSwitch/ConfigOnSwitch.ino#L46) for the reasons given [there](https://github.com/kentaylor/WiFiManager/blob/master/examples/ConfigOnSwitch/ConfigOnSwitch.ino#L46).

krzbor, 2016-10-14:
Thank you very much

closed 2018-01-26 by tablatronix