tzapu/WiFiManager #233: startConfigPortal seems not working

Slartitorto, 2016-10-07:
Hi, I'm in trouble with startConfigPortal.
Calling the startConfigPortal function, the new SSID and password seem not saved in eeprom; when the function exits, the esp links with old network with old password.
I tried with personal hotspot on my iphone: the esp go in configuration mode (by the startConfigPortal function), starts the AP and the captive portal, I connect on the new AP and the captive portal, then I set the iphone's personal hotspot, set the right password, but when esp.reset is execute or I restart the esp, I don't see any connection on the iphone, but I see the esp connecting to the old SSID.
Perhaps startConfigPortal save the new password only if the esp can't connect to the previuos SSID ?

kentaylor, 2016-10-07:
It works for others. Not a lot of clues in the problem description.

tablatronix, 2016-11-11:
Its possible your esp is not resetting by itself.

closed 2016-11-13 by Slartitorto