tzapu/WiFiManager #234: delete saved SSID and password

fubax, 2016-10-10:
Hi.
I use wifimanger for my project and it's works fine.now I need a way to delete the saved ssid and password  so I can use AP mode again and setup it with new ssid and pass.
I Know that if I turn off  wifi modem my esp goes to AP mode but I wish to connect a pushbutton to my esp8266 and clear the previous saved configuration with that pushbutton.
Please help me.
Thanks.

tzapu, 2016-10-10:
hi,

just call wifiManager.resetSettings() on button press, followd by and ESP.reset()

that should do it

cheers

fubax, 2016-10-10:
Thank you tzapu .But when I insert wifiManager.resetSettings() in my sketch it give me this error:
`exit status 1
'wifiManager' was not declared in this scope`
My wifimanager version is 0.12.0
I'm test it with WifiManager.resetSettings() but got the same error!

tzapu, 2016-10-10:
hi, sorry, it s probably not defined as a global.

do wifi.disconnect() in place of that followed by the rest.

cheers

fubax, 2016-10-10:
Thank you tzapu.
You save my day.
It works.

tzapu, 2016-10-10:
cool, i am glad.

have fun

closed 2016-10-10 by tzapu

kentaylor, 2016-10-10:
You could also consider a version of WiFiManager that operates on [button press only](https://github.com/kentaylor/WiFiManager#configuration-portal-initiation-not-automatic)