tzapu/WiFiManager #235: Switch between DHCP and static IP configuration

alexkirill, 2016-10-10:
I try to provide user ability to switch between DHCP and static IP configuration . 
How to do it?
I try to use setSTAStaticIPConfig in sketch with default values, but when, during the configuration on Captive Portal  leave it blank ( or enter wrong value) ESP successfully connects to WIFI as client but with wrong IP, DNS, etc. Hence device connected to wifi but not accessible from network due to wrong submask or gateway, ect. And it seems no way to go back to change them, or start device in AP mode again. Lot of words, hope I was plain))

kentaylor, 2016-10-10:
> hope I was plain

Not really but if I understand correctly you want the user to decide when the device is to be configured rather than autoconnect which throws up a config portal whenever it can not connect to WiFi. 

Assuming that is correct you want to initiate a configuration portal on a [button press](https://github.com/kentaylor/WiFiManager/#configuration-portal-initiation-not-automatic) rather than autoconnect.

alexkirill, 2016-10-10:
Not exactly, when user set up wifi parameters of his network via Captive Portal, he must have ability to choose use DHCP or set up IP/DNS/gate manually.

kentaylor, 2016-10-11:
Then perhaps [this example](https://github.com/tzapu/WiFiManager/blob/60c22d672fadfdcf139853f766d1824730944e13/examples/AutoConnectWithFSParametersAndCustomIP/AutoConnectWithFSParametersAndCustomIP.ino) is what you should copy. But when you don't want to set an IP make sure it doesn't call[ this line](https://github.com/tzapu/WiFiManager/blob/60c22d672fadfdcf139853f766d1824730944e13/examples/AutoConnectWithFSParametersAndCustomIP/AutoConnectWithFSParametersAndCustomIP.ino#L115). I don't use static IP but according to [this](https://github.com/esp8266/Arduino/issues/128#issuecomment-96456775) I think it will work.

alexkirill, 2016-10-11:
I using AutoConnectWithStaticIP.ino example, but it almost the same. But if I remove setSTAStaticIPConfig declaration it will be no possible to set static IP in Portal. I need the way to switch between DHCP and static assign in captive portal.

kentaylor, 2016-10-11:
> But if I remove setSTAStaticIPConfig declaration it will be no possible to set static IP in Portal.

Good point. 

You need to not make this [call](https://github.com/tzapu/WiFiManager/blob/60c22d672fadfdcf139853f766d1824730944e13/WiFiManager.cpp#L220) when you do not want a static IP. So you will need some logic change to achieve that.

alexkirill, 2016-10-11:
This logic must be implemented in captive portal and based on this library should to set wifi in dhcp or static mode. Library is pretty good but it weird that anyone haven't asked about it before.

alexkirill, 2016-10-11:
Ken, I test your version of WFmanager and it works like a charm. I customized template and library a bit as I want, but in general it better than this. & thanks for your fast replies

closed 2016-10-11 by alexkirill

kentaylor, 2016-10-12:
@alexkirill I'm glad you liked it. 

I'd be curious to see your modifications as I expect it would not have done exactly what you want as is and others may want the same functionality. The easiest for me would be fork my repository to your Github then commit your changes.

jindrichsirucek, 2018-07-12:
I was solving the same.. If you call 
`WiFi.config(...)`
and than you call `WiFi.begin(...);` - it useses always provided static conf.
So when you want to force ESP to use DHCP (after calling Wifi.config(...)) you have to call this inner function:
`wifi_station_dhcpc_start();`   
prior to 
`  WiFi.begin(...);`

tablatronix, 2018-07-12:
Yeah I think begin only resets dhcp if the config changes, I noted this elsewhere and kind of consider it a bug.

sajibkarmaker, 2018-11-04:
you don't need any kind off esp.reset or restart you just need to write a function where you should re define  IP,gateway as below to switch from fixed ip to dhcp(for esp8266 it will work definitely):
  function_name(){
IPAddress ip(0, 0, 0, 0);
  IPAddress subnet(0, 0, 0, 0);
  IPAddress gateway(0, 0, 0, 0);
  IPAddress dns(8, 8, 8, 8);
    IPAddress dn1(9, 9, 9, 9);
    WiFi.config(ip, gateway, subnet, dns, dn1);
    WiFi.begin(SSID,PSK);  //give your desired ssid 
}
according to the library if ip configured as 0.0.0.0 it switch from static ip to dhcp, and set fixed ip to do the reverse from dhcp to fixed ip.

Eusebio100, 2019-06-21:
I have the same problem, I can not get it to work in dhcp or static mode.
Can someone publish a code that works?
Thank.