tzapu/WiFiManager #236: Wifimanager and ArduinoOTA are not compatible

H4M3DR, 2016-10-12:
This is my code, it seems that arduinoOTA clears saved data from wifimanager and always starting in ap mode.

`#include <WiFiManager.h>
#include <ESP8266WiFi.h>
#include <DNSServer.h>   
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

void setup() {
  Serial.begin(9600);
  delay(10);
  
  WiFiManager wifiManager;
  wifiManager.autoConnect("Hamed");
  Serial.print("\nConnect\n");
  
   // Port defaults to 8266
   // ArduinoOTA.setPort(8266);

   // Hostname defaults to esp8266-[ChipID]
   ArduinoOTA.setHostname("myesp8266");

   // No authentication by default
   // ArduinoOTA.setPassword((const char *)"123");

   ArduinoOTA.onStart([]() {
     Serial.println("Start");
   });
   ArduinoOTA.onEnd([]() {
     Serial.println("\nEnd");
   });
   ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
     Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
   });
   ArduinoOTA.onError([](ota_error_t error) {
     Serial.printf("Error[%u]: ", error);
     if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
     else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
     else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
     else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
     else if (error == OTA_END_ERROR) Serial.println("End Failed");
   });
   ArduinoOTA.begin();
   Serial.println("Ready");
   Serial.print("IP address: ");
   Serial.println(WiFi.localIP());
  
}

void loop() {
   ArduinoOTA.handle();
}`

tzapu, 2016-10-12:
hi,

here s one of my sketches, with both wifimanger and arduinoota, maybe you can use as inspiration.

has worked fine for me over and over again.

cheers

malinengineer, 2016-10-13:
@tzapu Hi! You forgot to provide a link..

tzapu, 2016-10-13:
sorry about that
https://github.com/tzapu/SonoffBoilerplate/blob/master/SonoffBoilerplate.ino

good luck

closed 2016-11-03 by tzapu

drelephant, 2017-07-05:
Came across this via a search.

This gist is the most basic example that I could find that uses WifiManager and ArduinoOTA. I tried it and it worked perfectly.
https://gist.github.com/e857be405e4c77bdfec7b950c34f1b3f

KiLLeRRaT, 2018-06-06:
The above works, but when specifying a password using ArduinoOTA.setPassword("MyPassword"); things stop working.

tablatronix, 2018-06-07:
Since this issue contains no information of use please open a new one