tzapu/WiFiManager #238: set SSID in autoConnect() with name+MAC

mickeypop, 2016-10-14:
I am trying to get my Wemos D1 mini to set the AP name to Stand + the MAC address minus the semicolons, like **Stand5CCF7F238734**.

The GetMyMacAddress() function i wrote is clearly working, the serial out shows that.

Every time I try to pass a String or char variable to wifiManager.autoConnect() I get compiler errors. Even though the header file identifies String type the C file say const char.

if i pass macStr or `*`macStr  (char type)
&nbsp;&nbsp;&nbsp;**invalid conversion from 'char' to 'const char*' [-fpermissive]**

if i pass ap2 (String type) i get:
&nbsp;&nbsp;&nbsp;**no matching function for call to 'WiFiManager::autoConnect(String&)'**

My code;
```
#include `<ESP8266WiFi.h>` 
#include `<DNSServer.h>`
#include `<ESP8266WebServer.h>`
#include `<WiFiManager.h>` 

String ap = "Stand";
String ap2;
uint8_t mac[6];
char const macStr[19] = {0};

void setup() {
    Serial.begin(115200);
    WiFiManager wifiManager; 

    ap2 = ap + GetMyMacAddress();

    //std::string ap2;  // tried but more errors
    char *macStr = new char[ap2.length()+ 1 ];
    strcpy(macStr, ap2.c_str());

    wifiManager.autoConnect( "Stand" );       // works but not what i want
    //wifiManager.autoConnect( ap2 );         // cannot pass var to set SSID ???
    //wifiManager.autoConnect( *macStr );  // cannot pass var to set SSID ???
    //wifiManager.autoConnect( macStr );    // cannot pass var to set SSID ???

    Serial.println("connected...yeey :)");
    Serial.print("ap2"); Serial.print("    " ); Serial.print( ap2); Serial.println(" String");
    Serial.print("macStr"); Serial.print(" "); Serial.print( macStr ); Serial.println(" Char");
}

void loop() {   }

String GetMyMacAddress()
{   uint8_t mac[6];
    char macStr[18] = {0};
    WiFi.macAddress(mac);
    sprintf(macStr, "%02X%02X%02X%02X%02X%02X", mac[0],  mac[1], mac[2], mac[3], mac[4], mac[5]);
    return  String(macStr);
}
```
Serial out Returns: when connected
&nbsp;&nbsp; connected...yeey :)
&nbsp;&nbsp; ap2 &nbsp;&nbsp;&nbsp;      Stand5CCF7F238734 String
&nbsp;&nbsp; macStr Stand5CCF7F238734 Char

How can i pass the ap2 or macStr to set the AP SSID name?

kentaylor, 2016-10-16:
This [line](https://github.com/tzapu/WiFiManager/blob/fd15f986d3ab5831c543aadbd5cb07805f7b5c9f/WiFiManager.cpp#L127) and the following creates a string variable and calls autoconnect successfully so you could copy that construct. It is odd that the value of a const is changed but it is [possible](https://en.wikipedia.org/wiki/Const_(computer_programming)#Consequences). 

I'd recommend against using autoconnect for these [reasons](https://github.com/kentaylor/WiFiManager#configuration-portal-initiation-not-automatic). If you are happy with using the ChipID rather than the MAC address as your variable in the AP name then [this library](https://github.com/kentaylor/WiFiManager) already does that without modification.

tzapu, 2016-10-17:
Hi, try with `wifiManager.autoConnect( ap2.c_str() );`

Does that work? 
Cheers

mickeypop, 2016-10-17:
wifiManager.autoConnect( ap2.c_str() ); got it working.
i was not aware of the .c_str() operator   is has been a while since i needed and string minipulation.

iime to bone up i guess.

closed 2016-10-17 by mickeypop