tzapu/WiFiManager #241: ESP32 Compatability

forthlightning, 2016-11-03:
Thinking of porting this wonderful library over to the new ESP32 module? 

I'd be willing to contribute if you can point me in the right direction.

kentaylor, 2016-11-03:
I'm wondering how much pointing is required. The code is all here, just go for it. I'm fond of [this derivation](https://github.com/kentaylor/WiFiManager) for the reasons given there but you may not like it or prefer to roll your own.

tzapu, 2016-11-03:
as far as i can see webserver is not working yet on esp32 so this won't work either i'm afraid
https://github.com/espressif/arduino-esp32
got an esp32 in front of me that i didn t have a chance to play with yet, I do hope there will be some more progress on the arduino port as well

good luck and let us know if you get anywhere with it

forthlightning, 2016-11-03:
@kentaylor nice fork! Some of these are features I have been hoping for.

@tzapu thanks for the heads up about webserver, I'll poke around and see what there is to see.

Jorgen-VikingGod, 2017-01-05:
@tzapu webserver is working on current master. There are not all libs in yet - but I just try to use the webserver yesterday - it worked good.

seopyoon, 2017-02-03:
Does this work without the DNSServer.h?  Where can I find DNSServer.h for esp32?

tzapu, 2017-02-03:
it would work without dns server, code would need to be remove in the lib

cheers

seopyoon, 2017-02-03:
The WiFiServer.h provided by the arduino-esp32 repo, does not support `void on(args)` functions, so I failed to port it over =(

xinort, 2017-02-14:
=(

Curclamas, 2017-02-22:
So if I understand correctly we would need an arduino-esp32 equivalent of "ESP8266WebServer" https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer since "WiFiServer" ist not a full blown HTTP/Webserver. 
Does anybody of you know of any developments here?

Curclamas, 2017-03-10:
It looks like the issue I've referenced above has been resolved. So from what I understand it should now be possible to port (just copy it into the arduino esp32 libs folder) the webserver lib and cosequently the WiFiManager should work.

xinort, 2017-03-14:
has anyone tried this?

zarya, 2017-05-02:
After patching some of the includes to the ESP32 includes i was not able to get the ESP8266Webserver to be happy to much compile errors for me.

`In file included from lib/ESP8266WebServer/src/ESP8266WebServer.cpp:30:0:
lib/ESP8266WebServer/src/detail/RequestHandlersImpl.h: In constructor 'StaticRequestHandler::StaticRequestHandler(fs::FS&, const char*, const char*, const char*)':
lib/ESP8266WebServer/src/detail/RequestHandlersImpl.h:62:119: error: 'DEBUGV' was not declared in this scope
DEBUGV("StaticRequestHandler: path=%s uri=%s isFile=%d, cache_header=%s\r\n", path, uri, _isFile, cache_header);
^
lib/ESP8266WebServer/src/detail/RequestHandlersImpl.h: In member function 'virtual bool StaticRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String)':
lib/ESP8266WebServer/src/detail/RequestHandlersImpl.h:80:104: error: 'DEBUGV' was not declared in this scope
DEBUGV("StaticRequestHandler::handle: request=%s _uri=%s\r\n", requestUri.c_str(), _uri.c_str());
^
lib/ESP8266WebServer/src/ESP8266WebServer.cpp: In member function 'void ESP8266WebServer::close()':
lib/ESP8266WebServer/src/ESP8266WebServer.cpp:223:11: error: 'class WiFiServer' has no member named 'close'
_server.close();
^
In file included from lib/ESP8266WebServer/src/ESP8266WebServer.cpp:28:0:
lib/ESP8266WebServer/src/ESP8266WebServer.h: In instantiation of 'size_t ESP8266WebServer::streamFile(T&, const String&) [with T = fs::File; size_t = unsigned int]':
lib/ESP8266WebServer/src/detail/RequestHandlersImpl.h:111:41:   required from here
lib/ESP8266WebServer/src/ESP8266WebServer.h:132:60: error: no matching function for call to 'WiFiClient::write(fs::File&, int)'
return _currentClient.write(file, HTTP_DOWNLOAD_UNIT_SIZE);
^
In file included from /home/zarya/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiServer.h:24:0,
from lib/ESP8266WebServer/src/ESP8266WebServer.cpp:26:
/home/zarya/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.h:43:12: note: candidate: virtual size_t WiFiClient::write(uint8_t)
size_t write(uint8_t data);
^
/home/zarya/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.h:43:12: note:   candidate expects 1 argument, 2 provided
/home/zarya/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.h:44:12: note: candidate: virtual size_t WiFiClient::write(const uint8_t*, size_t)
size_t write(const uint8_t *buf, size_t size);
^
/home/zarya/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.h:44:12: note:   no known conversion for argument 1 from 'fs::File' to 'const uint8_t* {aka const unsigned char*}'
In file included from /home/zarya/.platformio/packages/framework-arduinoespressif32/cores/esp32/Stream.h:26:0,
from /home/zarya/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:150,
from lib/ESP8266WebServer/src/ESP8266WebServer.cpp:24:
/home/zarya/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:69:12: note: candidate: size_t Print::write(const char*, size_t)
size_t write(const char *buffer, size_t size)
^
/home/zarya/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:69:12: note:   no known conversion for argument 1 from 'fs::File' to 'const char*'
/home/zarya/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:61:12: note: candidate: size_t Print::write(const char*)
size_t write(const char *str)
^
/home/zarya/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:61:12: note:   candidate expects 1 argument, 2 provided
*** [.pioenvs/esp32/lib/ESP8266WebServer/ESP8266WebServer.o] Error 1
`

davidwallis3101, 2017-05-17:
Has anyone got this working with the esp32?

bbx10, 2017-05-27:
@tzapu There is discussion at espressif/arduino-esp32 on how best to merge WebServer and DNSServer libraries. Once that happens, it should be easier to port other libraries such as WiFiManager.

https://github.com/espressif/arduino-esp32/issues/395

bbx10, 2017-06-10:
Libraries for the adventurous. They are good enough for me to port my personal projects to ESP32 but there are TODOs, a patch, and an issue.

Be sure to use the esp32 branch.
https://github.com/bbx10/WiFiManager/tree/esp32

Patch required but not merged. Patch your local fork.
https://github.com/espressif/arduino-esp32/pull/428

Still a problem but the issue is currently closed.
https://github.com/espressif/arduino-esp32/issues/400

My personal projects are working on ESP32 with these libraries but your mileage may vary. The following are NOT supported by Espressif.
https://github.com/bbx10/WebServer_tng
https://github.com/bbx10/DNSServer_tng

yurykk, 2017-06-21:
Hi

I'm getting an error message when I'm trying to compile the example : 

Multiple libraries were found for "WiFi.h"
      Used: ..[censored]../Arduino/hardware/espressif/esp32/libraries/WiFi
     Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi

Please advise...

bbx10, 2017-06-21:
This is expected and not a problem. When there are include files with the same name in different directories the IDE reports which include it used to build the program. The correct one in this case is the ESP32 version.

yurykk, 2017-06-21:
but sketch will not compile because of this conflict....  Interesting, but I don't have this issue compiling ESP32 sample Web server sketch...

bbx10, 2017-06-21:
The compile is probably failing for some other reason. The following a successful compile which includes the multiple libraries warning but goes on to build the bin file. 

```
Multiple libraries were found for "WiFi.h"
 Used: /home/me/esp32/arduino-1.8.2/hardware/espressif/esp32/libraries/WiFi
 Not used: /home/me/esp32/arduino-1.8.2/libraries/WiFi
Using library WiFi at version 1.0 in folder: /home/me/esp32/arduino-1.8.2/hardware/espressif/esp32/libraries/WiFi 
Using library HTTPClient at version 1.1 in folder: /home/me/esp32/arduino-1.8.2/hardware/espressif/esp32/libraries/HTTPClient 
Using library ArduinoJson at version 5.10.0 in folder: /home/me/esp32/arduino-1.8.2/portable/sketchbook/libraries/ArduinoJson 
Using library WiFiClientSecure at version 1.0 in folder: /home/me/esp32/arduino-1.8.2/hardware/espressif/esp32/libraries/WiFiClientSecure 
Sketch uses 444851 bytes (33%) of program storage space. Maximum is 1310720 bytes.
Global variables use 32868 bytes (11%) of dynamic memory, leaving 262044 bytes for local variables. Maximum is 294912 bytes.
```

TheChapu, 2017-06-22:
WARNING: library DNSServer claims to run on [esp8266] architecture(s) and may be incompatible with your current board which runs on [esp32] architecture(s).
In file included from /tmp/arduino_modified_sketch_96123/sketch_jun22a.ino:12:0:
/opt/arduino-1.8.1/libraries/webserver/WebServer.h:131:37: error: conflicting declaration of C function 'long unsigned int millis()'
 extern "C" unsigned long millis(void);
                                     ^
In file included from /home/yo/Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:35:0,
                 from sketch/sketch_jun22a.ino.cpp:1:
/home/yo/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal.h:65:10: note: previous declaration 'uint32_t millis()'
 uint32_t millis();
          ^
In file included from /tmp/arduino_modified_sketch_96123/sketch_jun22a.ino:12:0:
/opt/arduino-1.8.1/libraries/webserver/WebServer.h: In member function 'int WebServer::read()':
/opt/arduino-1.8.1/libraries/webserver/WebServer.h:793:40: error: call of overloaded 'millis()' is ambiguous
     unsigned long timeoutTime = millis() + WEBDUINO_READ_TIMEOUT_IN_MS;
                                        ^
In file included from /home/yo/Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:35:0,
                 from sketch/sketch_jun22a.ino.cpp:1:
/home/yo/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal.h:65:10: note: candidate: uint32_t millis()
 uint32_t millis();
          ^
In file included from /tmp/arduino_modified_sketch_96123/sketch_jun22a.ino:12:0:
/opt/arduino-1.8.1/libraries/webserver/WebServer.h:131:26: note: candidate: long unsigned int millis()
 extern "C" unsigned long millis(void);
                          ^
/opt/arduino-1.8.1/libraries/webserver/WebServer.h:835:36: error: call of overloaded 'millis()' is ambiguous
         unsigned long now = millis();
                                    ^
In file included from /home/yo/Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:35:0,
                 from sketch/sketch_jun22a.ino.cpp:1:
/home/yo/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal.h:65:10: note: candidate: uint32_t millis()
 uint32_t millis();
          ^
In file included from /tmp/arduino_modified_sketch_96123/sketch_jun22a.ino:12:0:
/opt/arduino-1.8.1/libraries/webserver/WebServer.h:131:26: note: candidate: long unsigned int millis()
 extern "C" unsigned long millis(void);
                          ^
In file included from /opt/arduino-1.8.1/libraries/Ethernet/src/Ethernet.h:9:0,
                 from /opt/arduino-1.8.1/libraries/webserver/WebServer.h:32,
                 from /tmp/arduino_modified_sketch_96123/sketch_jun22a.ino:12:
/home/yo/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip/apps/dhcpserver.h: At global scope:
/opt/arduino-1.8.1/libraries/Ethernet/src/Dhcp.h:47:19: error: expected identifier before string constant
 #define HOST_NAME "WIZnet"

                   ^
/home/yo/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip/apps/dhcpserver.h:58:5: note: in expansion of macro 'HOST_NAME'
     HOST_NAME = 12,
     ^
/opt/arduino-1.8.1/libraries/Ethernet/src/Dhcp.h:47:19: error: expected '}' before string constant
 #define HOST_NAME "WIZnet"

                   ^
/home/yo/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip/apps/dhcpserver.h:58:5: note: in expansion of macro 'HOST_NAME'
     HOST_NAME = 12,
     ^
/opt/arduino-1.8.1/libraries/Ethernet/src/Dhcp.h:47:19: error: expected unqualified-id before string constant
 #define HOST_NAME "WIZnet"

                   ^
/home/yo/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip/apps/dhcpserver.h:58:5: note: in expansion of macro 'HOST_NAME'
     HOST_NAME = 12,
     ^
In file included from /home/yo/Arduino/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter/tcpip_adapter.h:48:0,
                 from /home/yo/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32/esp_event.h:23,
                 from /home/yo/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32/esp_wifi.h:68,
                 from /opt/arduino-1.8.1/libraries/WiFiManager-esp32/WiFiManager.h:32,
                 from /tmp/arduino_modified_sketch_96123/sketch_jun22a.ino:14:
/home/yo/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip/apps/dhcpserver.h:154:1: error: expected declaration before '}' token
 } dhcp_msg_option;
 ^
exit status 1
Error compiling for board ESP32 Dev Module.

bbx10, 2017-06-23:
Looks like you manually installed libraries in the wrong directory. The libraries may not be designed for ESP32. In arduino-1.8.1-linux64.tar.xz, I do not see a webserver library yet your log shows such a library.
`/opt/arduino-1.8.1/libraries/webserver/WebServer.h:` I suggest removing arduino1.8.1 directory and reinstall or upgrade to the latest. Install libraries in [sketch directory]/libraries/, not in the IDE libraries directory.

sglvladi, 2017-06-29:
@bbx10 You are amazing! Been waiting for this for months and you've done it all! 

A BIG shout out to everyone, @bbx10's solution {https://github.com/bbx10/WiFiManager/tree/esp32} works pretty much out of the box! All that is needed is to download his custom WebServer {https://github.com/bbx10/WebServer_tng} and DNSServer {https://github.com/bbx10/DNSServer_tng}, although I managed to run the AutoConnect example just with the first one. 

Pretty much identical functionality (as far as I have tested) with the original ESP8266 WiFiManager, but on the ESP32!!

Personally I think this should be shared as much as possible.  I am sure there are loads of people interested.

@bbx10 Why not turn the esp32 branch into a separate repo? It certainly worth it and would be great to also have some more details on the work you did to make it compatible.

bbx10, 2017-06-29:
@sglvladi Thanks, for the kind words! Ideally, my ESP32 code would be merged into this repo via PR so I am keeping it on a separate branch for now. There is [activity](https://github.com/espressif/arduino-esp32/issues/472) to create official Espressif repos for WebServer and, I hope, DNSServer and Hash libraries. This should make it easier for ESP8266 library maintainers to add support for ESP32.

yurykk, 2017-06-30:
@bbx10, I have some strange issue with WiFiManager on ESP32.

Many times it doesn't save configuration - it takes 3-4 times entering the password..

also, wifiManager.resetSettings(); doesn't work as expected. After reboot device still "remember" old settings...

Please advise...

Thanks

bbx10, 2017-06-30:
I have not had problems saving SSID/password. 

wifiManager.resetSettings() does not work because of problem with WiFi.disconnect(true) not working. Still waiting for a fix for this. I found if WiFi.disconnect(true) is called after successfully connecting, it does erase the saved SSID/password.

miknuc, 2017-07-02:
@bbx10 
I tried your fork. Seems to compile :)
I am just wondering how to use it. I was about to move a project from esp8266 to esp32 and ran into an issue with persisting the wifi configuration:
The examples look quite like the esp8266 counterparts and make of use of SPIFFS which is still WIP for esp32 to my knowledge. I do not understand how they could work with esp32. 
Am I missing something?

Michael

bbx10, 2017-07-02:
@miknuc SPIFFS is not available on ESP32. For configuration information, I use the ESP32 Preferences library. Similary, if an app uses the ESP8266 EEPROM library, it can be converted to use Preferences.

programmer131, 2017-07-04:
i've removed FS related functions from library and esp8266webserver works fine with esp32. will update wifimanager library to see if it works.

miknuc, 2017-07-04:
Thank you for the suggestion. It's even simpler to persist these values using Preferences instead of SPIFFS/JSON.

elecherf, 2017-07-10:
Hi, looks the @bbx10 links https://github.com/bbx10/WiFiManager/tree/esp32 have disapeared. Plz share. Thank you

bbx10, 2017-07-10:
@elecherf The link and repo are working fine.

LongmontRobotics, 2017-07-13:
I have installed the three zip files into Arduino IDE 1.8.2:
https://github.com/bbx10/WiFiManager/tree/esp32
https://github.com/bbx10/WebServer_tng
https://github.com/bbx10/DNSServer_tng

I get this error:
C:\Users\Donald\Documents\Arduino\libraries\WiFiManager-esp32\WiFiManager.cpp:746:84: error: 'class WiFiClient' has no member named 'localIP'

     server->sendHeader("Location", String("http://") + toStringIp(server->client().localIP()), true);

Please help locate this missing member.

miknuc, 2017-07-13:
This function is not yet part of the expressif libraries. You need to apply the patch mentioned by bbx10 on Jun 10 somewhere in the expressif libs, see patch file for details. 

Btw: I am new to git and do not know how to takeover the patch with a simple git command, so I applied the small change manually. There is probably a better way to do this. Any hint how the standard procedures look like is appreciated. 
Michael

LongmontRobotics, 2017-07-13:
I don't understand what this all means.
What patch are you referring to ?

LongmontRobotics, 2017-07-13:
OK, figured it out.

Thanks

bbx10, 2017-07-13:
@LongmontRobotics  @miknuc The patch is no longer required because the fix was merged with the latest version from https://github.com/espressif/arduino-esp32. Just get the latest files.

LongmontRobotics, 2017-07-13:
If I Add .ZIP library from the Sketch-> Include Library, will that over right the existing arduino-esp32 directories ?

bbx10, 2017-07-13:
arduino-esp32 is not a library so I do not think that will work.

Depending on your OS, find the step that does `git clone https://github.com/espressif/arduino-esp32.git esp32`. Change to the same directory then do `git pull` to get the latest updates. `git clone` is only needed on the initial installation.

https://github.com/espressif/arduino-esp32#installation-instructions

LongmontRobotics, 2017-07-13:
I just figured that out.
Thanks for your help.

amattice, 2017-08-01:
@elecherf, the code is there, but you are probably tying to use the click shortcuts which incorrectly place a trailing parentheses.  Try these...

https://github.com/bbx10/WiFiManager/tree/esp32

https://github.com/bbx10/WebServer_tng

https://github.com/bbx10/DNSServer_tng

unixb0y, 2017-08-03:
Is there a good "ESP8266WebServer" alternative yet?
I'm new to ESP32 development and that WiFiServer lacks a whole lot of stuff. Today I tried to get URL parameters like you could with the old board using server.args() but no chance here.

gjt211, 2017-08-23:
Hi,
Is there a way to include both the original wifimanager (that I use for my ESP8266 projects) and this one for my ESP32 projects on a single computer/arduino install?
It would require a different name for this library I guess. Is there a simple way to change that?

sukeshak, 2017-08-24:
I found a port here for ESP32... anyone tested yet?
https://github.com/zhouhan0126/WIFIMANAGER-ESP32

Also changes done for these two as well.
WebServer https://github.com/zhouhan0126/WebServer-esp32
DNSServer https://github.com/zhouhan0126/DNSServer---esp32

gjt211, 2017-09-25:
@bbx10 
I have been trying to work out a way to get `wifiManager.resetSettings()` working.
I seen a comment about using `make` from the command line to erase settings, but this is not practical for boards that I don't physically have access to.

It seems the settings are stored using preferences, and there are commands available in the ESP32 Arduino core to modify the preferences. I was wondering where to look in your code to find where preferences is done?
I have a quick look but can't find anything. I was hoping I could add some code to fix/add this as I really need to be able to have `wifiManager.resetSettings()` working.

bbx10, 2017-09-25:
The problem is in an ESP32 core function so you found the correct place.

https://github.com/espressif/arduino-esp32/issues/400

Scott--R, 2017-09-26:
@sukeshak I tried the ESP32 ported version and it seems to work but apparently the wifi persist is not yet implemented on the ESP32, so every time I power cycle the device it forgets the SSID and PWD. Looks like we need to implement a workaround using the preferences library until the wifi persist function / feature is done.

sukeshak, 2017-09-26:
@Scott--R  I think the preferences API is for Arduino side and not for ESP-IDF. 
I need something under ESP-IDF

sukeshak, 2017-09-26:
I think it would be SPIFF?
https://github.com/espressif/esp-idf/tree/master/examples/storage/spiffs

Testato, 2017-09-26:
Also in arduino-esp32 now there is the SPIFF library

https://github.com/espressif/arduino-esp32/tree/master/libraries/SPIFFS

sukeshak, 2017-09-26:
I just kicked off a discussion to understand what is the best option
https://esp32.com/viewtopic.php?f=13&t=3146&p=14743#p14743

Testato, 2017-09-26:
we have also EEPROM library for value storing

sukeshak, 2017-09-27:
What about the version from @bbx10 above?
I haven't tried it... I need something for ESP-IDF side.

eirikso, 2018-01-03:
I am currently setting up several ESP32 where I need a WiFiManager. So I have started testing tis port: https://github.com/zhouhan0126/WIFIMANAGER-ESP32

It works pretty OK. I have done two edits to make it work properly:

**1. A change in WiFiAP.cpp so that the captive portal pops up automatically** (it is not very user friendly to have people entering 192.168.4.1 in their browser). This edit fixed it on all the devices I have tested. The problem is described here: https://github.com/zhouhan0126/WIFIMANAGER-ESP32/issues/2

You simply paste these three lines from line 95 in WiFiAP.cpp:
 
tcpip_adapter_dns_info_t dns_info;
dns_info.ip.u_addr.ip4.addr = IPAddress(192,168,4,1);
tcpip_adapter_set_dns_info(TCPIP_ADAPTER_IF_AP, TCPIP_ADAPTER_DNS_MAIN, &dns_info);

**2. A change in WiFiManager.cpp to make sure that the ESP32 connects automatically** after the user have entered SSID and PWD. Before this change I had to reset the ESP32 after entering the credentials.

Add this at line 216: 
ESP.restart();
delay(1000);

This problem is described here: https://github.com/zhouhan0126/WIFIMANAGER-ESP32/issues/1

With these two edits it seems like WIFIMANAGER-ESP32 works good. My only problem is the fact that it is impossible to delete the credentials before a connection is OK. I would very much like to be able to set up the ESP32 so that there is a "factory reset" option at startup.

Does anyone here know if there is a fix for that now?

eirikso, 2018-01-03:
I don't know if this helps anyone, but it is the same for these:

Serial.println( WiFi.SSID() );
Serial.println( WiFi.psk() );

They return rubbish if called before a connection is made and return what they are supposed to do if called after the connection is made.

Testato, 2018-01-03:
Yhanks for your tips, may you create a pull request  ?

For the connection problem there is many thinks that work only after a connection, often is enought start a connection to a fake user+password value and after you can continue

eirikso, 2018-01-03:
Unfortunately, after more testing I am still not satisfied with the stability of this solution. The ESP32 fails to connect way too often. So I guess I'll just have to wait until someone that is much better than me at programming solves this properly. :-)

Until then I'll use my old trusted ESP8266 for this.

tablatronix, 2018-01-03:
These problems sound like esp32 lib bugs

eirikso, 2018-01-04:
Yes. It probably is. The main problem is that it way too often starts the config portal even if I know that it had stored the credentials for my wifi that has a 100% signal. And I can not find a reliable source of the error. When I disconnect and connect power again it is completely random if it connects or not.

jwktje, 2018-01-22:
@eirikso What would be your best advice at this point? I've been developing an ESP32 based solution that needs WifiManager and found this Github issue after looking around for a solution. Just stick with ESP8266 for now?

tablatronix, 2018-01-22:
I might start working on this, i think i saw a pr but i do not see a working fork, i might start from scratch. If anyone can gather the changes needed

[81 more replies]

closed 2021-06-21 by tablatronix