tzapu/WiFiManager #243: Access SSID and Password

dudesky71, 2016-11-11:
Hello,

I'm trying to use WiFi Manager with ADAFRUIT IO with an ESP8266.

Your code works fine by itself. My problem is passing the saved WiFi SSID and PW to Adafruit's SSID and PW code. It defines it as a constant string.

Is there an easy way to do this?

Thanks!

tzapu, 2016-11-11:
hi,

fetch it from WiFi.SSID() and WIFI.psk() after the setup is done

cheers
alex

dudesky71, 2016-11-11:
Sorry, don't mean to be a bonehead but their code does this...

#define WIFI_SSID               "SSID goes here"
#define WIFI_PASS              "PW goes here"

These sit in a config.h file

If I remove the quotes and insert WiFi.SSID() and WIFI.psk() It tells me they are not declared in the scope. I'm sure i'm doing it wrong... Any ideas? Much appreciated :)

tablatronix, 2016-11-11:
You should be able to just remove adafruit connection code and just use wifimanager autoconenct

dudesky71, 2016-11-14:
I got this to work but it seems to not play nicely with Adafruit.IO
When WiFimanager is in the code, it randomly locks up the esp8266 when sending sensor data to Adafruit.IO. If I remove WiFimanager from the startup, it works fine. Everything is running latest. Any ideas?
@tzapu

tablatronix, 2016-11-14:
basic sketch example ?

dudesky71, 2016-11-14:
#include <DHT.h>
#include <SPI.h>
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <WiFiManager.h>
#include "AdafruitIO_WiFi.h"

/************************ Example Starts Here *******************************/

#define IO_USERNAME    "xxxxxx"
#define IO_KEY         "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

#define WIFI_SSID      ""
#define WIFI_PASS      ""

#define DHTPIN 4     // what digital pin we're connected to. Mapped to pin D2 on WeMOS
#define DHTTYPE DHT11   // DHT 11
//#define BLYNK_PRINT Serial    // Comment this out to disable prints and save space
#define ANALOG_PIN A0  // analog pin 0

// Initialize DHT sensor.
DHT dht(DHTPIN, DHTTYPE);

AdafruitIO_WiFi io(IO_USERNAME, IO_KEY, WIFI_SSID, WIFI_PASS);

// analog state
int currentwetness = 0;
int lastwetness = -1;
int tempint = 0;
int humint = 0;
unsigned long previousMillis = 0;
const long interval = 10000;

// set up the 'analog' feed
AdafruitIO_Feed *wetnessfeed = io.feed("wetness");

// set up the 'DHT' feed float
AdafruitIO_Feed *tempfeed = io.feed("temp");
AdafruitIO_Feed *humfeed = io.feed("hum");

// set up the 'DHT' feed integer
AdafruitIO_Feed *tempfeedint = io.feed("tempint");
AdafruitIO_Feed *humfeedint = io.feed("humint");

void setup() {

  // start the serial connection
  Serial.begin(9600);
    Serial.println("");
    Serial.println(WiFi.SSID());
    Serial.println(WiFi.psk());
// put your setup code here, to run once:

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    //wifiManager.autoConnect("Test-DongleAP");
    //or use this for auto generated name ESP + ChipID
    wifiManager.autoConnect();

[... 111 lines cut ...]
  Serial.println(currentwetness);
  wetnessfeed->save(currentwetness);
  delay(1000);
   Serial.print("Sending -> ");
   Serial.println(temp);
  String tempstring = String(temp);
  tempfeed->save(temp);
  delay(1000);
  tempfeedint->save(tempint);
    Serial.print("Sending -> ");
    delay(1000);
  Serial.println(hum);
   humfeed->save(hum);
   delay(1000);
   humfeedint->save(humint);
    Serial.println("Sent...");
    Serial.println("");
  // Blink the blue LED once for data sent confirmation
  digitalWrite(2, LOW);   // sets the LED on
  delay(100);
  digitalWrite(2, HIGH);   // sets the LED off
    // wait ten seconds (10000 milliseconds == 10 seconds)
    Serial.println("Pausing 10 Seconds...");
   }
  }

dudesky71, 2016-11-14:
`#include <DHT.h>
#include <SPI.h>
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <WiFiManager.h>
#include "AdafruitIO_WiFi.h"

/************************ Example Starts Here *******************************/

#define IO_USERNAME    "xxxxxx"
#define IO_KEY         "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

#define WIFI_SSID      ""
#define WIFI_PASS      ""

#define DHTPIN 4     // what digital pin we're connected to. Mapped to pin D2 on WeMOS
#define DHTTYPE DHT11   // DHT 11
//#define BLYNK_PRINT Serial    // Comment this out to disable prints and save space
#define ANALOG_PIN A0  // analog pin 0

// Initialize DHT sensor.
DHT dht(DHTPIN, DHTTYPE);

AdafruitIO_WiFi io(IO_USERNAME, IO_KEY, WIFI_SSID, WIFI_PASS);

// analog state
int currentwetness = 0;
int lastwetness = -1;
int tempint = 0;
int humint = 0;
unsigned long previousMillis = 0;
const long interval = 10000;

// set up the 'analog' feed
AdafruitIO_Feed *wetnessfeed = io.feed("wetness");

// set up the 'DHT' feed float
AdafruitIO_Feed *tempfeed = io.feed("temp");
AdafruitIO_Feed *humfeed = io.feed("hum");

// set up the 'DHT' feed integer
AdafruitIO_Feed *tempfeedint = io.feed("tempint");
AdafruitIO_Feed *humfeedint = io.feed("humint");

void setup() {

  // start the serial connection
  Serial.begin(9600);
    Serial.println("");
    Serial.println(WiFi.SSID());
    Serial.println(WiFi.psk());
// put your setup code here, to run once:

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    //wifiManager.autoConnect("Test-DongleAP");
    //or use this for auto generated name ESP + ChipID
    wifiManager.autoConnect();

[... 112 lines cut ...]
  wetnessfeed->save(currentwetness);
  delay(1000);
   Serial.print("Sending -> ");
   Serial.println(temp);
  String tempstring = String(temp);
  tempfeed->save(temp);
  delay(1000);
  tempfeedint->save(tempint);
    Serial.print("Sending -> ");
    delay(1000);
  Serial.println(hum);
   humfeed->save(hum);
   delay(1000);
   humfeedint->save(humint);
    Serial.println("Sent...");
    Serial.println("");
  // Blink the blue LED once for data sent confirmation
  digitalWrite(2, LOW);   // sets the LED on
  delay(100);
  digitalWrite(2, HIGH);   // sets the LED off
    // wait ten seconds (10000 milliseconds == 10 seconds)
    Serial.println("Pausing 10 Seconds...");
   }
  }
`

dudesky71, 2016-11-14:
//#include <DHT.h>
//#include <SPI.h>
//#include <ESP8266WiFi.h>
//#include <DNSServer.h>
//#include <WiFiManager.h>
//#include "AdafruitIO_WiFi.h"

/************************ Example Starts Here *******************************/

#define IO_USERNAME    "xxxxxx"
#define IO_KEY         "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

#define WIFI_SSID      ""
#define WIFI_PASS      ""

#define DHTPIN 4     // what digital pin we're connected to. Mapped to pin D2 on WeMOS
#define DHTTYPE DHT11   // DHT 11
//#define BLYNK_PRINT Serial    // Comment this out to disable prints and save space
#define ANALOG_PIN A0  // analog pin 0

// Initialize DHT sensor.
DHT dht(DHTPIN, DHTTYPE);

AdafruitIO_WiFi io(IO_USERNAME, IO_KEY, WIFI_SSID, WIFI_PASS);

// analog state
int currentwetness = 0;
int lastwetness = -1;
int tempint = 0;
int humint = 0;
unsigned long previousMillis = 0;
const long interval = 10000;

// set up the 'analog' feed
AdafruitIO_Feed *wetnessfeed = io.feed("wetness");

// set up the 'DHT' feed float
AdafruitIO_Feed *tempfeed = io.feed("temp");
AdafruitIO_Feed *humfeed = io.feed("hum");

// set up the 'DHT' feed integer
AdafruitIO_Feed *tempfeedint = io.feed("tempint");
AdafruitIO_Feed *humfeedint = io.feed("humint");

void setup() {

  // start the serial connection
  Serial.begin(9600);
    Serial.println("");
    Serial.println(WiFi.SSID());
    Serial.println(WiFi.psk());
// put your setup code here, to run once:

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    //wifiManager.autoConnect("Test-DongleAP");
    //or use this for auto generated name ESP + ChipID
    wifiManager.autoConnect();

[... 111 lines cut ...]
  Serial.println(currentwetness);
  wetnessfeed->save(currentwetness);
  delay(1000);
   Serial.print("Sending -> ");
   Serial.println(temp);
  String tempstring = String(temp);
  tempfeed->save(temp);
  delay(1000);
  tempfeedint->save(tempint);
    Serial.print("Sending -> ");
    delay(1000);
  Serial.println(hum);
   humfeed->save(hum);
   delay(1000);
   humfeedint->save(humint);
    Serial.println("Sent...");
    Serial.println("");
  // Blink the blue LED once for data sent confirmation
  digitalWrite(2, LOW);   // sets the LED on
  delay(100);
  digitalWrite(2, HIGH);   // sets the LED off
    // wait ten seconds (10000 milliseconds == 10 seconds)
    Serial.println("Pausing 10 Seconds...");
   }
  }

dudesky71, 2016-11-14:
ok sorry... What's the secret to pasting code so it all shows up? The forum is stripping some things out :/

tzapu, 2016-11-14:
add three backticks ` before and three  after the code

tablatronix, 2016-11-14:
single tick is for inline code, triple ticks for blocks

dudesky71, 2016-11-14:
'''#include <DHT.h>
#include <SPI.h>
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <WiFiManager.h>
#include "AdafruitIO_WiFi.h"

/************************ Example Starts Here *******************************/

#define IO_USERNAME    "xxxxxx"
#define IO_KEY         "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

#define WIFI_SSID      ""
#define WIFI_PASS      ""

#define DHTPIN 4     // what digital pin we're connected to. Mapped to pin D2 on WeMOS
#define DHTTYPE DHT11   // DHT 11
//#define BLYNK_PRINT Serial    // Comment this out to disable prints and save space
#define ANALOG_PIN A0  // analog pin 0

// Initialize DHT sensor.
DHT dht(DHTPIN, DHTTYPE);

AdafruitIO_WiFi io(IO_USERNAME, IO_KEY, WIFI_SSID, WIFI_PASS);

// analog state
int currentwetness = 0;
int lastwetness = -1;
int tempint = 0;
int humint = 0;
unsigned long previousMillis = 0;
const long interval = 10000;

// set up the 'analog' feed
AdafruitIO_Feed *wetnessfeed = io.feed("wetness");

// set up the 'DHT' feed float
AdafruitIO_Feed *tempfeed = io.feed("temp");
AdafruitIO_Feed *humfeed = io.feed("hum");

// set up the 'DHT' feed integer
AdafruitIO_Feed *tempfeedint = io.feed("tempint");
AdafruitIO_Feed *humfeedint = io.feed("humint");

void setup() {

  // start the serial connection
  Serial.begin(9600);
    Serial.println("");
    Serial.println(WiFi.SSID());
    Serial.println(WiFi.psk());
// put your setup code here, to run once:

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    //wifiManager.autoConnect("Test-DongleAP");
    //or use this for auto generated name ESP + ChipID
    wifiManager.autoConnect();

[... 113 lines cut ...]
  wetnessfeed->save(currentwetness);
  delay(1000);
   Serial.print("Sending -> ");
   Serial.println(temp);
  String tempstring = String(temp);
  tempfeed->save(temp);
  delay(1000);
  tempfeedint->save(tempint);
    Serial.print("Sending -> ");
    delay(1000);
  Serial.println(hum);
   humfeed->save(hum);
   delay(1000);
   humfeedint->save(humint);
    Serial.println("Sent...");
    Serial.println("");
  // Blink the blue LED once for data sent confirmation
  digitalWrite(2, LOW);   // sets the LED on
  delay(100);
  digitalWrite(2, HIGH);   // sets the LED off
    // wait ten seconds (10000 milliseconds == 10 seconds)
    Serial.println("Pausing 10 Seconds...");
   }
  }
'''

dudesky71, 2016-11-14:
<pre>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">DHT</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><b><font color="#d35400">SPI</font></b><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">ESP8266WiFi</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">DNSServer</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">WiFiManager</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">"AdafruitIO_WiFi.h"</font>

<font color="#95a5a6">/************************ Example Starts Here *******************************/</font>

<font color="#5e6d03">#define</font> <font color="#000000">IO_USERNAME</font> &nbsp;&nbsp;&nbsp;<font color="#005c5f">"xxxxxx"</font>
<font color="#5e6d03">#define</font> <font color="#000000">IO_KEY</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">"xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"</font>

<font color="#5e6d03">#define</font> <font color="#000000">WIFI_SSID</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">""</font>
<font color="#5e6d03">#define</font> <font color="#000000">WIFI_PASS</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">""</font>

<font color="#5e6d03">#define</font> <font color="#000000">DHTPIN</font> <font color="#000000">4</font> &nbsp;&nbsp;&nbsp;&nbsp;<font color="#434f54">// what digital pin we're connected to. Mapped to pin D2 on WeMOS</font>
<font color="#5e6d03">#define</font> <font color="#000000">DHTTYPE</font> <font color="#000000">DHT11</font> &nbsp;&nbsp;<font color="#434f54">// DHT 11</font>
<font color="#434f54">//#define BLYNK_PRINT Serial &nbsp;&nbsp;&nbsp;// Comment this out to disable prints and save space</font>
<font color="#5e6d03">#define</font> <font color="#000000">ANALOG_PIN</font> <font color="#000000">A0</font> &nbsp;<font color="#434f54">// analog pin 0</font>

<font color="#434f54">// Initialize DHT sensor.</font>
<font color="#000000">DHT</font> <font color="#000000">dht</font><font color="#000000">(</font><font color="#000000">DHTPIN</font><font color="#434f54">,</font> <font color="#000000">DHTTYPE</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#000000">AdafruitIO_WiFi</font> <font color="#000000">io</font><font color="#000000">(</font><font color="#000000">IO_USERNAME</font><font color="#434f54">,</font> <font color="#000000">IO_KEY</font><font color="#434f54">,</font> <font color="#000000">WIFI_SSID</font><font color="#434f54">,</font> <font color="#000000">WIFI_PASS</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">// analog state</font>
<font color="#00979c">int</font> <font color="#000000">currentwetness</font> <font color="#434f54">=</font> <font color="#000000">0</font><font color="#000000">;</font>
<font color="#00979c">int</font> <font color="#000000">lastwetness</font> <font color="#434f54">=</font> <font color="#434f54">-</font><font color="#000000">1</font><font color="#000000">;</font>
<font color="#00979c">int</font> <font color="#000000">tempint</font> <font color="#434f54">=</font> <font color="#000000">0</font><font color="#000000">;</font>
<font color="#00979c">int</font> <font color="#000000">humint</font> <font color="#434f54">=</font> <font color="#000000">0</font><font color="#000000">;</font>
<font color="#00979c">unsigned</font> <font color="#00979c">long</font> <font color="#000000">previousMillis</font> <font color="#434f54">=</font> <font color="#000000">0</font><font color="#000000">;</font>
<font color="#00979c">const</font> <font color="#00979c">long</font> <font color="#000000">interval</font> <font color="#434f54">=</font> <font color="#000000">10000</font><font color="#000000">;</font>

<font color="#434f54">// set up the 'analog' feed</font>
<font color="#000000">AdafruitIO_Feed</font> <font color="#434f54">*</font><font color="#000000">wetnessfeed</font> <font color="#434f54">=</font> <font color="#000000">io</font><font color="#434f54">.</font><font color="#000000">feed</font><font color="#000000">(</font><font color="#005c5f">"wetness"</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#434f54">// set up the 'DHT' feed float</font>
<font color="#000000">AdafruitIO_Feed</font> <font color="#434f54">*</font><font color="#000000">tempfeed</font> <font color="#434f54">=</font> <font color="#000000">io</font><font color="#434f54">.</font><font color="#000000">feed</font><font color="#000000">(</font><font color="#005c5f">"temp"</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">AdafruitIO_Feed</font> <font color="#434f54">*</font><font color="#000000">humfeed</font> <font color="#434f54">=</font> <font color="#000000">io</font><font color="#434f54">.</font><font color="#000000">feed</font><font color="#000000">(</font><font color="#005c5f">" [... cut ...] ="#434f54">// sets the LED off</font>
 &nbsp;&nbsp;&nbsp;<font color="#434f54">// wait ten seconds (10000 milliseconds == 10 seconds)</font>
 &nbsp;&nbsp;&nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">"Pausing 10 Seconds..."</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;<font color="#000000">}</font>
 &nbsp;<font color="#000000">}</font>

</pre>

dudesky71, 2016-11-14:
any ideas @tzapu ?
Thank You for looking.

tablatronix, 2016-11-14:
I would suggest reducing this to a minimum amount of code to reproduce, there is just too much going on, it makes for a bad test. Start pulling everything out but the 2 things being tested

dudesky71, 2016-11-14:
this is a super simple sketch. There is hardly anything going on as-is. There are 3 data pushes to Adafruit's server. That is all. Anything lees in the code and it will not be doing anything...

tablatronix, 2016-11-14:
well no
This would be a simple sketch
or use the examples and add to them, you have spi dht libraries etc.

```
#include <ESP8266WiFi.h>
#include <WiFiManager.h>
#include "AdafruitIO_WiFi.h"

/************************ Example Starts Here *******************************/

#define IO_USERNAME    "xxxxxx"
#define IO_KEY         "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

#define WIFI_SSID      ""
#define WIFI_PASS      ""

AdafruitIO_WiFi io(IO_USERNAME, IO_KEY, WIFI_SSID, WIFI_PASS);

AdafruitIO_Feed *testfeed = io.feed("test");

void setup() {
  Serial.begin(115200);
  Serial.println("");

  WiFiManager wifiManager;
  wifiManager.autoConnect();

  WiFi.printDiag(Serial);

  // connect to io.adafruit.com
  Serial.println("");
  Serial.print("Connecting to Adafruit IO");
  io.connect();

  // wait for a connection
  while(io.status() < AIO_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  Serial.print("adafruit io connected");
  Serial.println(io.statusText());
}

void loop() {
  io.run();
  testfeed->save(millis());
  delay(2000);
}
```

Either way i doubt this is going to work with wifimanager, since this library handled wifi for you, you might have to use the adafruit mqtt library directly.

dudesky71, 2016-11-14:
ok thank you

dudesky71, 2016-11-16:
I just tried another WeMos module. I've tried a HUZZAH. I've tried the basic MQTT client and they all lock up or reset the module within minutes of running. Please see the log and dump below... Can you help with this? 

Thank You

Sending counter val 248...OK!

Sending counter val 249...OK!

Sending counter val 250...OK!

Sending counter val 251...OK!

Sending counter val 252...OK!

Sending counter val 253...OK!

Sending counter val 254...OK!

Sending counter val 255...OK!

Sending counter val 256...OK!

Sending counter val 257...OK!

Sending counter val 258...Failed
Connecting to MQTT... Connection failed
Retrying MQTT connection in 5 seconds...
Connection failed
Retrying MQTT connection in 5 seconds...

MQTT Connected!

Sending counter val 259...OK!

Sending counter val 260...Failed
Connecting to MQTT... 
MQTT Connected!

Sending counter val 261...Failed
Connecting to MQTT... 
MQTT Connected!

Sending counter val 262...Failed
Connecting to MQTT... Connection failed
Retrying MQTT connection in 5 seconds...

MQTT Connected!

Sending counter val 263...OK!

Sending counter val 264...OK!
Connecting to MQTT... Connection failed
Retrying MQTT connection in 5 seconds...
Connection failed
Retrying MQTT connection in 5 seconds...
Connection failed
Retrying MQTT connection in 5 seconds...

Soft WDT reset

ctx: cont 
sp: 3ffef710 end: 3ffef900 offset: 01b0

>>>stack>>>
3ffef8c0:  00000001 3ffee638 3ffee814 40201e32  
3ffef8d0:  3fffdad0 00000000 3ffee8c8 40201e88  
3ffef8e0:  3fffdad0 00000000 3ffee8c8 40203844  
3ffef8f0:  feefeffe feefeffe 3ffee8e0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

tzapu, 2016-11-16:
that looks more like an error related to mqtt
there is an esp8266 exception decoder floating around somehwre you can install and see exactly where the error came from
good luck

tablatronix, 2016-11-16:
not sure, i am using the adafruit mqtt anon_time example , it works fine.

Its not an exception, soft wdt reset usually means something is restarting it

tablatronix, 2016-11-16:
usually a infinite loop tripping the wathcdog. check for rogue while loops without a delay or yield()

dudesky71, 2016-11-16:
here is the basic sketch...
`include "ESP8266WiFi.h"
include "Adafruit_MQTT.h"
include "Adafruit_MQTT_Client.h"

/************************* WiFi Access Point *********************************/

#define WLAN_SSID       "...your SSID..."
#define WLAN_PASS       "...your password..."

/************************* Adafruit.io Setup *********************************/

#define AIO_SERVER      "io.adafruit.com"
#define AIO_SERVERPORT  1883                   // use 8883 for SSL
#define AIO_USERNAME    "...your AIO username (see https://accounts.adafruit.com)..."
#define AIO_KEY         "...your AIO key..."

/************ Global State (you don't need to change this!) ******************/

// Create an ESP8266 WiFiClient class to connect to the MQTT server.
WiFiClient client;
// or... use WiFiFlientSecure for SSL
//WiFiClientSecure client;

// Setup the MQTT client class by passing in the WiFi client and MQTT server and login details.
Adafruit_MQTT_Client mqtt(&client, AIO_SERVER, AIO_SERVERPORT, AIO_USERNAME, AIO_KEY);

/****************************** Feeds ***************************************/

// Setup a feed called 'photocell' for publishing.
// Notice MQTT paths for AIO follow the form: <username>/feeds/<feedname>
Adafruit_MQTT_Publish photocell = Adafruit_MQTT_Publish(&mqtt, AIO_USERNAME "/feeds/photocell");

// Setup a feed called 'onoff' for subscribing to changes.
Adafruit_MQTT_Subscribe onoffbutton = Adafruit_MQTT_Subscribe(&mqtt, AIO_USERNAME "/feeds/onoff");

/*************************** Sketch Code ************************************/

// Bug workaround for Arduino 1.6.6, it seems to need a function declaration
// for some reason (only affects ESP8266, likely an arduino-builder bug).
void MQTT_connect();

void setup() {
  Serial.begin(115200);
  delay(10);

  Serial.println(F("Adafruit MQTT demo"));

  // Connect to WiFi access point.
  Serial.println(); Serial.println();
  Serial.print("Connecting to ");
  Serial.println(WLAN_SSID);

  WiFi.begin(WLAN_SSID, WLAN_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();

  Serial.println("WiFi connected");
  Serial.println("IP address: "); Serial.println(WiFi.localIP());

  // Setup MQTT subscription for onoff feed.
  mqtt.subscribe(&onoffbutton);
}

uint32_t x=0;

void loop() {
  // Ensure the connection to the MQTT server is alive (this will make the first
  // connection and automatically reconnect when disconnected).  See the MQTT_connect
  // function definition further below.
  MQTT_connect();

  // this is our 'wait for incoming subscription packets' busy subloop
  // try to spend your time here

  Adafruit_MQTT_Subscribe *subscription;
  while ((subscription = mqtt.readSubscription(5000))) {
    if (subscription == &onoffbutton) {
      Serial.print(F("Got: "));
      Serial.println((char *)onoffbutton.lastread);
    }
  }

  // Now we can publish stuff!
  Serial.print(F("\nSending photocell val "));
  Serial.print(x);
  Serial.print("...");
  if (! photocell.publish(x++)) {
    Serial.println(F("Failed"));
  } else {
    Serial.println(F("OK!"));
  }

  // ping the server to keep the mqtt connection alive
  // NOT required if you are publishing once every KEEPALIVE seconds
  /*
  if(! mqtt.ping()) {
    mqtt.disconnect();
  }
  */
}

// Function to connect and reconnect as necessary to the MQTT server.
// Should be called in the loop function and it will take care if connecting.
void MQTT_connect() {
  int8_t ret;

  // Stop if already connected.
  if (mqtt.connected()) {
    return;
  }

  Serial.print("Connecting to MQTT... ");

  uint8_t retries = 3;
  while ((ret = mqtt.connect()) != 0) { // connect will return 0 for connected
       Serial.println(mqtt.connectErrorString(ret));
       Serial.println("Retrying MQTT connection in 5 seconds...");
       mqtt.disconnect();
       delay(5000);  // wait 5 seconds
       retries--;
       if (retries == 0) {
         // basically die and wait for WDT to reset me
         while (1);
       }
  }
  Serial.println("MQTT Connected!");`

tzapu, 2016-11-16:
this
```
Soft WDT reset

ctx: cont 
sp: 3ffef710 end: 3ffef900 offset: 01b0

stack>>>
3ffef8c0: 00000001 3ffee638 3ffee814 40201e32

3ffef8d0: 3fffdad0 00000000 3ffee8c8 40201e88

3ffef8e0: 3fffdad0 00000000 3ffee8c8 40203844

3ffef8f0: feefeffe feefeffe 3ffee8e0 40100718

<<<stack<<<

ets Jan 8 2013,rst cause:2, boot mode:(1,7)

ets Jan 8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```
looks like an watchdog triggered reset, so there is a problem, not just a reset

tablatronix, 2016-11-16:
I am gonna guess this
So your mqtt is timing out or failing and this is resetting your board.

```
       if (retries == 0) {
         // basically die and wait for WDT to reset me
        Serial.println("MQTT retry FAIL, rebooting."); // <--- add this to debug
         while (1);
       }
```

dudesky71, 2016-11-16:
I just changed over to a completely different communication method used by thinger.io

Here is the example sketch. It is super simple. It still locks up the 8266 modules. Seems like its a core 8266 problem...?

<pre>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">ESP8266WiFi</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">ThingerESP8266</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#define</font> <font color="#000000">USERNAME</font> <font color="#005c5f">"your_user_name"</font>
<font color="#5e6d03">#define</font> <font color="#000000">DEVICE_ID</font> <font color="#005c5f">"your_device_id"</font>
<font color="#5e6d03">#define</font> <font color="#000000">DEVICE_CREDENTIAL</font> <font color="#005c5f">"your_device_credential"</font>

<font color="#5e6d03">#define</font> <font color="#d35400">SSID</font> <font color="#005c5f">"your_wifi_ssid"</font>
<font color="#5e6d03">#define</font> <font color="#000000">SSID_PASSWORD</font> <font color="#005c5f">"your_wifi_ssid_password"</font>

<font color="#000000">ThingerESP8266</font> <font color="#000000">thing</font><font color="#000000">(</font><font color="#000000">USERNAME</font><font color="#434f54">,</font> <font color="#000000">DEVICE_ID</font><font color="#434f54">,</font> <font color="#000000">DEVICE_CREDENTIAL</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<font color="#d35400">pinMode</font><font color="#000000">(</font><font color="#000000">BUILTIN_LED</font><font color="#434f54">,</font> <font color="#00979c">OUTPUT</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#000000">thing</font><font color="#434f54">.</font><font color="#000000">add_wifi</font><font color="#000000">(</font><font color="#d35400">SSID</font><font color="#434f54">,</font> <font color="#000000">SSID_PASSWORD</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">// digital pin control example (i.e. turning on/off a light, a relay, configuring a parameter, etc)</font>
 &nbsp;<font color="#000000">thing</font><font color="#000000">[</font><font color="#005c5f">"led"</font><font color="#000000">]</font> <font color="#434f54">&lt;&lt;</font> <font color="#000000">digitalPin</font><font color="#000000">(</font><font color="#000000">BUILTIN_LED</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">// resource output example (i.e. reading a sensor value)</font>
 &nbsp;<font color="#000000">thing</font><font color="#000000">[</font><font color="#005c5f">"millis"</font><font color="#000000">]</font> <font color="#434f54">&gt;&gt;</font> <font color="#000000">outputValue</font><font color="#000000">(</font><font color="#d35400">millis</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#434f54">// more details at </font><u><font color="#434f54">http://docs.thinger.io/arduino/</font></u><font color="#434f54"></font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<font color="#000000">thing</font><font color="#434f54">.</font><font color="#000000">handle</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

</pre>

tablatronix, 2016-11-16:
Are you running wifimanager all the time ?

dudesky71, 2016-11-16:
Yes. Is there a way to shut it down after the credentials get saved?

tablatronix, 2016-11-16:
I was testing with this

```
/***************************************************
  Adafruit MQTT Library ESP8266 Adafruit IO Anonymous Time Query

  Must use the latest version of ESP8266 Arduino from:
    https://github.com/esp8266/Arduino

  Works great with Adafruit's Huzzah ESP board & Feather
  ----> https://www.adafruit.com/product/2471
  ----> https://www.adafruit.com/products/2821

  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Adafruit Industries.
  MIT license, all text above must be included in any redistribution
 ****************************************************/
#include <ESP8266WiFi.h>
#include <WiFiManager.h>
#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"

/************************* WiFi Access Point *********************************/
#define WLAN_SSID       "wlanmtsu"
#define WLAN_PASS       ""

/************************* Adafruit.io Setup *********************************/
#define AIO_SERVER      "io.adafruit.com"
#define AIO_SERVERPORT  8883

WiFiClientSecure client;
Adafruit_MQTT_Client mqtt(&client, AIO_SERVER, AIO_SERVERPORT);

Adafruit_MQTT_Subscribe timefeed = Adafruit_MQTT_Subscribe(&mqtt, "time/seconds");

// set timezone offset from UTC
int timeZone = -5; // UTC - 4 eastern daylight time (nyc)
int interval = 4; // trigger every X hours

int last_min = -1;

void timecallback(uint32_t current) {
  // adjust to local time zone
  current += (timeZone * 60 * 60);
  int curr_hour = (current / 60 / 60) % 24;
  int curr_min  = (current / 60 ) % 60;
  int curr_sec  = (current) % 60;

  Serial.print("Time: "); 
  Serial.print(curr_hour); Serial.print(':'); 
  Serial.print(curr_min); Serial.print(':'); 
  Serial.println(curr_sec);
  
  // only trigger on minute change
  if(curr_min != last_min) {
    last_min = curr_min;
    
    Serial.println("This will print out every minute!");
    digitalWrite(BUILTIN_LED, LOW);
    delay(500);
    digitalWrite(BUILTIN_LED, HIGH);    
  }
}

void setup() {

  pinMode(BUILTIN_LED,OUTPUT);

  Serial.begin(115200);
  delay(10);
  Serial.println("");
  Serial.print(F("\nAdafruit IO anonymous Time Demo"));
  
  WiFi.printDiag(Serial);

  if(WiFi.SSID() == "") {
    WiFiManager wifiManager;
    wifiManager.autoConnect();
  }

  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    delay(500);
    Serial.print(F("."));
  }
  Serial.println(F(" WiFi connected."));

  timefeed.setCallback(timecallback);
  mqtt.subscribe(&timefeed);

}

void loop() {

  // Ensure the connection to the MQTT server is alive (this will make the first
  // connection and automatically reconnect when disconnected).  See the MQTT_connect
  // function definition further below.
  MQTT_connect();

  // wait 10 seconds for subscription messages
  // since we have no other tasks in this example.
  mqtt.processPackets(10000);

  // keep the connection alive
  mqtt.ping();

}

// Function to connect and reconnect as necessary to the MQTT server.
// Should be called in the loop function and it will take care if connecting.
void MQTT_connect() {
  int8_t ret;

  // Stop if already connected.
  if (mqtt.connected()) {
    return;
  }

  Serial.print("Connecting to MQTT... ");

  uint8_t retries = 3;
  while ((ret = mqtt.connect()) != 0) { // connect will return 0 for connected
       Serial.println(mqtt.connectErrorString(ret));
       Serial.println("Retrying MQTT connection in 5 seconds...");
       mqtt.disconnect();
       delay(5000);  // wait 5 seconds
       retries--;
       if (retries == 0) {
         // basically die and wait for WDT to reset me
        Serial.println("MQTT out of retries, restating ESP");
         while (1);
       }
  }

  Serial.println("MQTT Connected!");
}
```

dudesky71, 2016-11-17:
I can see the AP drop offline so I don't think it is that.

closed 2018-01-26 by tablatronix

wyojustin, 2018-02-12:
I had the same question using the 8266 Feather HUZZAH.  I finally just bit the bullet and hacked the AdafruitIO_ESP8266.cpp code.  One small change makes the code work.

1. Include WiFi Mananger header file near the top of "AdafruitIO_ESP8266.cpp"
`
#ifdef ESP8266
#include "AdafruitIO_ESP8266.h"
#include <WiFiManager.h> // TJS: Hacked in for Mac's Klok

2. Change _connect to read as below

```
void AdafruitIO_ESP8266::_connect()
{
   // TJS: Use WiFi Manager hack
  // ### hacked in for Mac's kock... and it worked!!!
  WiFiManager wifiManager;
  //wifiManager.resetSettings();// uncomment to forget previous wifi manager settings
  wifiManager.autoConnect("KLOK");
  Serial.println("yay connected");
/* // original code
  delay(100);
  WiFi.begin(_ssid, _pass);
  delay(100);
  _status = AIO_NET_DISCONNECTED;
*/
}
```

elanozturk, 2018-06-13:
i find out the way,thank you.
```
  Serial.println(WiFi.SSID());
  Serial.println(WiFi.psk());
```

gabi68, 2019-11-06:
Also,
Serial.printf("BSSID: %s\n", WiFi.BSSIDstr().c_str());
Good luck!!!

preface-egeriekung, 2022-06-11:
> It tells me they are not declared in the scope. I'm sure i'm doing it wrong... Any ideas? Much appreciated :)
Maybe try

```
#define WIFI_SSID WiFi.SSID()
#define WIFI_PASS WiFi.psk()
```