tzapu/WiFiManager #245: WiFi credentials entered in captive portal are not remanent upon ESP reset

croisez, 2016-11-18:
Hi,
I use WiFiManager 0.12.0, and esp8266 Arduino board add-on version 2.3.0.
I use a generic esp8266 module (an ESP12-F).

I want to use WiFiManager as fallback when the AP is OFF.
After having entered my SSID and pwd in the captive portal, the ESP switches in STA mode and succeeds in connecting to my AP.

But after that, whenever I reset the ESP (by forcing the RESET to zero), I expect the previously entered credentials to be used, but no. The ESP fails in STA mode and go back to captive portal.

Here is the traces I see on screen, on reboot after the RESET:

*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: SET AP STA
...

What is the meaning of Connection result: 0??
Why?

croisez, 2016-11-18:
On the first connection, I get a successful connection. Here are the WM traces:

*WM: Handle root
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Waiting for connection result with time out
*WM: Connection result: 
*WM: 3

Connection result is equal to 3 in that case, not 0.

kentaylor, 2016-11-18:
`WiFi.persistent(true)
`
is the default, only specify it in the unlikely event you want 
`WiFi.persistent(false)
`

@croisez asks
> What is the meaning of Connection result: 0??

0 corresponds to [WL_IDLE_STATUS](https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/ESP8266WiFi/src/include/wl_definitions.h) . WL_IDLE_STATUS is an [Arduino term](https://www.arduino.cc/en/Reference/WiFiStatus) and the ESP8266 doesn't exactly match the definition because the number of attempts does not expire in this case. It will wait forever until it sees the AP then attempt to connect.

I guess the 31 means no signal detected? The negative RSSI values are dBm which in your case show a [low signal strength](http://www.metageek.com/training/resources/understanding-rssi.html) suggesting if you were closer to the router you would see different behaviour. 

You provide a good example of why I argue that [autoconnect is bad](https://github.com/kentaylor/WiFiManager#configuration-portal-initiation-not-automatic) . There is  discussion on [future changes to autoconnect](https://github.com/tzapu/WiFiManager/pull/157#issuecomment-260169950).

SensorsIot, 2016-11-19:
I have a very similar behavior. I use webupdate and print the ssit and psk before I autoconnect. Even when the ssid/psk are displayed right, it enters wuit often in the AP mode. setConnectTimeout(60) did not change this behavior as it enters the AP mode quite quickly.

croisez, 2016-11-21:
Yes @SensorsIot, I also had the situation where the connection could not be done, while the printf of WiFi.SSID() and WiFi.psk() show good values.
But most of the time, printing WiFi.SSID() and WiFi.psk() show empty values.
I suppose that a call to WiFi.begin(ssid,psk) should do the saving-to-eeprom process, and that normally, upon reboot, a call to WiFi.begin() without args should use previously saved credentials, and get success with it.
Is it right?

kentaylor, 2016-11-21:
@croisez asks

> Is it right?

Yes it is and calling WiFi.begin() after reboot is not useful.

When @croisez says

> But most of the time, printing WiFi.SSID() and WiFi.psk() show empty values.

It sounds like the [same issue](https://github.com/tzapu/WiFiManager/issues/247) as @hutch120.

hutch120, 2017-02-12:
We are making some progress on this issue over at https://github.com/tzapu/WiFiManager/issues/247

closed 2018-01-26 by tablatronix