tzapu/WiFiManager #248: WiFiManager (KT version) traps if no networks found in scan

hutch120, 2016-11-21:
@kentaylor I'm not sure if I'm supposed to report this here given your version is a fork?? Anyway, couldn't see a way to mark this issue to your repo.

Your version traps if wifi scan finds no networks.

In the WiFiManager destructor that runs free(networkIndices), the pointer is uninitalised if there were no networks found during the scan.

To fix, initalise the pointer in WiFiManager.h
```
int *networkIndices     = NULL;
```

http://www.cplusplus.com/reference/cstdlib/free/

Cheers,
Simon

tzapu, 2016-11-21:
feel free to post all stuff here, i see @kentaylor is monitoring issues quite closely

kentaylor, 2016-11-21:
Thanks @hutch120. I never would have spotted that. Feel free to do a pull request if you spot other stuff.  Pull requests work in a fork.

Also thanks @tzapu, you can't have issues in a fork so it is discuss here or nowhere. I've been watching of late but there are long periods where I don't.

closed 2017-08-31 by hutch120

tzapu, 2017-08-31:
hi guys, just wanted to mention it s fine to report here, jsut make sure it's obvious is related to ken's fork
and keep up the good work
cheers