tzapu/WiFiManager #253: Maximum Url Size for WiFi Manager

pieman64, 2016-11-27:
@tzapu and @kentaylor I think I may have hit the 2K character url limit with WiFi Manager.

I have 2 regular sized parameters (Blynk token and server IP address) and then ideally I need 14 (or more) large parameters (1000 characters each).

With regular parameters plus two large parameters the system appears to work (probably just under the 2048 character limit imposed by many browsers). When I go above this the system fails.

It could be the json buffer that is overflowing when I have more than 2 large parameters but I suspect it's the web browser character limit.

I might try adding a separate file to the Filesystem as an addition to config.json but then it will not be available to WiFiManager.

Any ideas how I might get WiFiManager to accept circa 14000 characters?

pieman64, 2016-11-27:
A separate file for the Filesystem is fine with > 14000 characters which confirms the json buffer is not the problem.

kentaylor, 2016-11-28:
@pieman, I think you are asking the wrong question. Do you really want to want to collect 14 parameters of 1000 characters each on a minimalist device? What happens when you want to replace the device? Are you going to enter them all again? What about if you have a fleet of them? Surely you want to be able to set the parameters all through the same interface so you can copy and paste when you set up a new device and just change the few that are specific to that instance and surely you want to manage the whole fleet all through the one interface. 

If that is the case you need the [IOT Configurator](http://configure.urremote.com/).

pieman64, 2016-11-28:
@kentaylor the ESP creates the 1000 character parameters so it's not really a case of entering them manually. The ESP can also email the parameters so then it becomes a simple cut and paste exercise.
Having previously used Nano's then I don't consider 16M ESP's to be minimalist devices.

I have stumbled across references to the IOT Configurator from time to time so I will take a look when I have a minute.

Having set up the second file outside of WiFi Manager I might just go with 2 large parameters in WiFi Manager and the rest in the second file. As it is at the moment the fleet reference is not really an issue as we hard code the 14000 characters into our sketch and then users change them to suit their requirements but it is all done automatically by the ESP.

kentaylor, 2016-11-28:
@pieman64 I'd agree ESP's are amazingly powerful devices for just a few dollars which is what makes them game changing. I meant minimalist in the sense of user interface and the way users are likely to interact with them. The focus is not on the individual device. Even phones, which have a decent user interface, are that way now. You get a new one and it sets itself up with all the same apps and configuration as the previous one and if you have several, they mostly work seamlessly together.

I'm confident cloud based configuration will become de rigueur in IOT too.

pieman64, 2016-11-29:
@kentaylor I registered with IOT Configurator and even though I have a basic understanding of SQL and phpMyAdmin I didn't get very far.

When I try to AddDevice it is giving a 500 server error. Is the server working OK?

kentaylor, 2016-12-01:
Hmmm, @pieman64 that's no good. Do you mean you get a 500 server error when using phpMyAdmin? I can't reproduce your problem. We can discuss this issue further on the [IOT Configurator discussion](https://gitter.im/IOTConfigurator/Lobby#).

cosmocracy, 2016-12-20:
In case you need more ideas, consider having some (minimal) javascript in the config portal send the parameters individually via AJAX.  We'd need to tweak the server-side code to handle this but it would allow each parameter to be sent individually, perhaps bypassing the limit you're encountering trying to send them all in the same request.

closed 2018-01-26 by tablatronix