tzapu/WiFiManager #258: SSID & password sent unsecured

pratikpanchal22, 2016-12-04:
Configuring the network from the captive portal sends the SSID and APPSK as a GET request to the server running on ESP unsecured. The SSID and APPSK are seen in the url:

`http://192.168.4.1/wifisave?s=HomeWifi&p=homeWifi402754
`

This is a security flaw and allows the snoopers to hijack the WiFi credentials very easily.

What would be a feasible solution for that?

cosmocracy, 2016-12-20:
In this case the HTTPS server would have to be run on the ESP8266.  The ESP8266 barely supports HTTPS as a client--even then using older TLSv1.1--so I doubt you'll see this capability in the future on the ESP8266 platform.  Better to open an issue at the [ESP8266 project](https://github.com/esp8266/Arduino) instead of here.

For reference, see this issue (among the others in the ESP8266 project): https://github.com/esp8266/Arduino/issues/2733

I think this issue is a "non-issue" for WiFiManager for the following reasons:
* The vulnerability only exists for the small duration of time that the ESP8266 is in AP mode
* You're unlikely to have anyone else connected to the ESP8266 AP during your config session
* What you're asking for wouldn't work the way you want since you wouldn't have an SSL certificate on the ESP8266--your browser would give you a DANGER--CANNOT VERIFY SITE error when you try to configure (assuming you managed to get a self-signed cert to be served out by the ESP8266)

tablatronix, 2016-12-29:
It should at least be a post so it is not stored in access logs. Someone made a pull request to do this already ideallly allow both.

suculent, 2017-04-03:
Yes, this should be a POST from OWASP point of view, regardless transport security.

closed 2018-01-26 by tablatronix