tzapu/WiFiManager #262: ESP8266 Electrodragon module

daniftodi, 2016-12-11:
Hi,

I'm trying to use WifiManager with ESP8266 on electrodragon module (http://www.electrodragon.com/product/wifi-iot-relay-board-based-esp8266/).

I just uploaded AutoConnect example, connected to 220V without serial connected to my PC because of strage noise ( I think because of poor quality of AC filter on module ), and waited for Wifi AP to appear on my phone/pc.

Nothing happens, I can't see any feedback from device.

Tell me please some sugestions about how I can debug this module and use WifiManager for it ?

cosmocracy, 2016-12-20:
Sounds like your best avenue of troubleshooting is to investigate why you can't use your serial connection for debug messages.  If serial isn't working because of noise, wouldn't you think that would impact the ESP8266's function, too? Just suggesting...

daniftodi, 2016-12-20:
My device works as Wifi client and connects without problems to other AP.
I would like to use WifiManager to simply the configuration process.

cosmocracy, 2016-12-20:
Yes. But if serial doesn't work why would you think other functions would reliably? My experience and intuition tell me you would be best served focusing on getting serial debug working so you can troubleshoot using the debug messages. Otherwise you're wandering in the dark and no one will be able to give you advice (other than to buy another [inexpensive] ESP8266).

closed 2018-01-26 by tablatronix