tzapu/WiFiManager #263: Is it possible to reconfigure after autoconnect?

smadds, 2016-12-12:
This is such a great library. I cannot thank you enough for the time it has saved.

Can I ask how to force a reconfigure after using autoconnect? I read in your docs:
> _Instead of calling autoConnect() which does all the connecting and failover configuration portal setup for you, you need to use startConfigPortal(). Do not use BOTH._

I would like to give users the opportunity to reconfigure some of the parameters (including the additional ones I have created), even if the unit has connected to wifi. I'd like to look for a continuous button press for a few seconds to trigger the config mode.

kentaylor, 2016-12-13:
In my view @smadds  would be better off to use the [IOT configurator](http://configure.urremote.com/) for setting parameters other than WiFi configuration.

Here is an example of configuration on [button press](https://github.com/kentaylor/WiFiManager/tree/master/examples/ConfigOnSwitch) and here is [an example](https://github.com/kentaylor/esp8266-weather-station/blob/master/examples/WeatherDHT22ThreeScreensWiFiManager/WeatherDHT22ThreeScreensWiFiManager.ino) where the button must be pushed for a few seconds to respond, and in this case you see a sliding bar on a screen.  A button library is used that detects short press, long press, double press and, is button pressed now.

smadds, 2016-12-13:
Thanks, Ken. I had a look at IOT Configurator and it looks interesting - albeit putting a dependency for a 3rd-party system. My main problem is not the button, but how to reconfigure the WiFi settings within WM. The best I've come up with so far is to use the WM reset function and then restart the device. This is not ideal, however, as it clears all of the settings - but works as a "restore to factory defaults" type of use case.

kentaylor, 2016-12-13:
When @smadds says:

> but how to reconfigure the WiFi settings within WM. 

I don't understand but that's fine.

tablatronix, 2016-12-29:
Theres an example called ondemand why not just do that or handle the button press how ever you want to start it. Whats the problem?

jaygrooms, 2017-01-03:
@smadds Did you ever figure out your button press to factory defaults?  I am TOTALLY new to programming and have been beating my head against the wall for about a week trying to figure it out.

Any help you could give would be great!

Tnx
-J

tzapu, 2017-01-03:
guys, here s an example that includes logic for a button triggered reset of settings
https://github.com/tzapu/SonoffBoilerplate/blob/master/SonoffBoilerplate.ino

tablatronix, 2017-02-08:
or

```C++

#define TRIGGER_PIN 0

void setup(){
  pinMode(TRIGGER_PIN, INPUT_PULLUP);
}

//crappy button handler for testing
//erase config reboot
void handleButton(){
  int debounce = 50;
  if ( digitalRead(TRIGGER_PIN) == LOW ){
    delay(debounce);
    if(digitalRead(TRIGGER_PIN) == LOW ){
      WiFiManager wifiManager;      
      wifiManager.resetSettings();
      delay(1000);
      ESP.reset();
    }
  }
}

void loop() {
  handleButton();
}
```

smadds, 2017-02-08:
My request was to make the web page reset button more protected by adding a
confirm stage, which Theo has kindly done.

There is a button sequence for clearing locally stored settings - hold
button down for more than 4 seconds. See wiki/button page.

One issue I constantly have is a new install tries to connect to Theo's
mqtt server and this has a very long timeout, during which the system is
unresponsive. Beware if this!!

I usually try to get it into web manager mode first (4 button presses) ,
set the log on to its Wi-Fi access point, then configure the client Wi-Fi
ssid & password. I then look for it showing up on my Dhcp server, go to the
web page and update the mqtt before anything else. Lots of reboots, but
once you are at this stage it's plain sailing.

tablatronix, 2017-02-09:
Interesting

closed 2018-01-26 by tablatronix

rodri16, 2018-08-02:
Hi I was wondering how to implement an wifiManager.autoConnect() on Setup and a wifiManager.startConfigPortal() when the button is pressed.
Both work very well separated but together ESP8266 crashes.
The problem with OnDemand example is that ESP8266 doesn't connect from scratch.
I need the device to get connected when it is plugged if the router is available and have the possibility to reconfigure with the button

tablatronix, 2018-08-02:
Just add autoconnect and either 2 wm instances or move the class obj outside of setup

tablatronix, 2018-08-02:
This exact thing is discussed in another issue i think

tzapu, 2018-08-02:
if WiFi.SSID() is an empty string then you could call autoConnect() and on button you call startConfigPortal()
so in all other cases WiFiManager would be not started.
on most of my sketches i just use autoConnect with setTimeout and a restart on said timeout. if router is unavailable in a few mins, it reboots and tries connecting again. works pretty well (for me)

rodri16, 2018-08-09:
Great I made it work, both with Master and Dev branch. The problem is that with Dev branch after WiFi configuration I cannot connect to a server client to send data or to update the clock using NTP server.
Same code runs ok with Master branch and connects to servers...

tablatronix, 2018-08-09:
Can you elaborate, are you in sta mode or sta + ap mode, 
Also what does the debug say ?

rodri16, 2018-08-09:
After doing ap mode and WiFi configuration, in sta mode.

Dev Branch. Problem trying to connect to servers
[image]
Master, connected to NTP and data servers
[image]

rodri16, 2018-08-09:
I checked and in Master, after pressing the button, startConfigPortal crashes after scan networks
In dev it works but never get connection to servers.

tablatronix, 2018-08-09:
can you add an print mode after exit to see which mode you are in

or add WiFi.mode(WIFI_STA); and see if it starts working ?

tablatronix, 2018-08-09:
( that crash in master is new also, hmm )

which esp lib ver are you using ?

tablatronix, 2018-08-09:
#688 
similar issue

tablatronix, 2018-08-09:
investigsating..

tablatronix, 2018-08-09:
hmm I was trying to make a quick test sketch, but I cannot get this ntp example to work at all.
know a good ping lib ?

tablatronix, 2018-08-09:
ah i forgot to start udp nm

tablatronix, 2018-08-09:
cannot reproduce on staging either

rodri16, 2018-08-10:
I am using this library https://github.com/gmag11/NtpClient

rodri16, 2018-08-10:
It works adding WiFi.mode(WIFI_STA);!!!

tablatronix, 2018-08-10:
What version of esp lib? There is clearly a bug

rodri16, 2018-08-10:
SDK Version 2.2.1
Core version 2.4.1
Boot version 4

rodri16, 2018-08-10:
Also in Dev branch setAPStaticIPConfig does not work! It always goes to 192.168.4.1

*WM: [1] SETTINGS ERASED 
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: TimeZone
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [3] WIFI station disconnect 
*WM: [1] No saved credentials, skipping wifi 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] xxxxxx
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  xxxxxx
*WM: [1] AP IP address: 192.168.4.1
Entered config mode
192.168.4.1

//gets called when WiFiManager enters configuration mode
void configModeCallback (WiFiManager *myWiFiManager) {
  Serial.println("Entered config mode");
  myWiFiManager->setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0)); 
  Serial.println(WiFi.softAPIP());
  //if you used auto generated SSID, print it
  Serial.println(myWiFiManager->getConfigPortalSSID());
  //entered config mode, make led toggle faster
  ticker.attach(0.2, tick);
}

tablatronix, 2018-08-10:
thanks, callback was moved
5ee1ae78c0fc808ad5eacec398d19a658ec35aa3

tablatronix, 2018-08-10:
I have adde some debugging  for this can you test and post debug log ?

rodri16, 2018-08-28:
*WM: [2] AccessPoint set password is VALID 
*WM: [1] BioBarica 
*WM: [3] WiFi station enable 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  xxxxxx
*WM: [1] Custom AP IP/GW/Subnet: 
*WM: [0] [ERROR] softAPConfig failed! 
*WM: [1] AP IP address: 192.168.4.1
Entered config mode
192.168.4.1
xxxxxx
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan done in 1582ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] NUM CLIENTS: 0

tablatronix, 2018-08-28:
Are you still using callback? Did you try setting it before wm starts?

rodri16, 2018-09-03:
Yes using callback! Added settings but still going to 192.168.4.1

tablatronix, 2018-09-03:
#720

rodri16, 2018-09-03:
Updated development branch but still...
*WM: [2] AccessPoint set password is VALID 
*WM: [1] XXXXX
*WM: [3] WiFi station enable 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  XXXXX
Entered config mode
0.0.0.0
XXXXX
*WM: [1] Custom AP IP/GW/Subnet: 
*WM: [0] [ERROR] softAPConfig failed! 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] Portal Timeout In 118 seconds
*WM: [2] WiFi Scan ASYNC completed in 1589 ms
*WM: [2] WiFi Scan ASYNC found: 5
*WM: [2] Portal Timeout In 88 seconds

rodri16, 2018-09-03:
Doing a wifiManager.resetSettings();

mounted file system
reading config file
opened config file
{"mqtt_server":"xxxx","mqtt_port":"80","TimeZone":"-3","ip":"192.168.1.198","gateway":"192.168.1.254","subnet":"255.255.255.0"}
parsed json
xxxx
-3
*WM: [1] SETTINGS ERASED 
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: TimeZone
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [3] WIFI station disconnect 
*WM: [1] No saved credentials, skipping wifi 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] xxxx
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  xxxx
*WM: [1] Custom AP IP/GW/Subnet: 
*WM: [1] AP IP address: 10.0.1.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  10.0.1.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] WiFi Scan ASYNC completed in 2186 ms
*WM: [2] WiFi Scan ASYNC found: 9

And then pressing the button to start config portal
*WM: [2] AccessPoint set password is VALID 
*WM: [1] xxxx 
*WM: [3] WiFi station enable 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  xxxx
*WM: [1] Custom AP IP/GW/Subnet: 
*WM: [0] [ERROR] softAPConfig failed! 
*WM: [1] AP IP address: 10.0.1.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  10.0.1.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] Portal Timeout In 118 seconds
*WM: [2] WiFi Scan ASYNC completed in 1589 ms
*WM: [2] WiFi Scan ASYNC found: 8

Now it is working with 10.0.1.1

tablatronix, 2018-09-03:
Ok well there is no callback if already connected so probably always set config first

rodri16, 2018-09-03:
Even adding setAPStaticIPConfig first

tablatronix, 2018-09-03:
same output ?

rodri16, 2018-09-04:
Yes same output, even if I delete callback and add configuration first.

tablatronix, 2018-09-04:
Ok sorry this is ap sta config , so different issue, ill try to debug today

tablatronix, 2018-09-04:
@rodri16 can you create a new issue for this and fill out the info, I want to track this desperately this is getting confusing. You clearly have "softAPConfig failed" so this is either a sketch issue or a esp bug or wrong mode in wm before setting config.

guysoft, 2019-05-06:
Implementation using the double reset library:
https://github.com/guysoft/Mqtt_Wifi_manager