tzapu/WiFiManager #264: Is there an easy way to change wifi channels?

gregsvo, 2016-12-13:
I've got a project I'm working on at my office, but there are many, many wireless devices in the building. The problem (to my knowledge) is that **the ESP8266 defaults to channel 1**, which is heavily used in this office- and is causing connectivity issues. **I would like to change the ESP8266's channel to something else.** Is there a way in the configuration in WifiManager to do so?

**Thank you!** I'm very new at Arduino/ESP8266, so I appreciate your patience.

mushfiq-cse05-buet, 2016-12-14:
u can use the WiFi functions of arduino core see the documentation Thanks

gregsvo, 2016-12-14:
I was hoping for a more informational answer than that. I've already seen the documentation, but wifi manager didn't seem to be respecting my changes.

ggriesse, 2017-03-15:
Also looking for a why to change this ..

closed 2018-01-26 by tablatronix

girisan, 2019-10-06:
> Also looking for a why to change this ..

I can give a 'why' for an option to specify channel...
In my case, when the device goes into config mode, the SSID is not visible in the scanned list, Resetting the device repeatedly didnt help. The probable cause was, that the channel probably clashed with many other hotspots in the neighborhood. A simple work around is to let the device specify the channel, which it can randomize each time it gets into the config mode.
At present, in such cases, WiFiManager will be entirely ineffective.

tablatronix, 2019-10-06:
#936