tzapu/WiFiManager #265: IP Address

sergionaf, 2016-12-16:
Is possible to show the IP Address assigned to ESP from the DHCP server of the connected wireless network ?

Humancell, 2016-12-29:
I don't believe this has anything to do with the WiFi Manager library ... you can always use:

Serial.println(WiFi.localIP());

skx, 2017-02-01:
Agreed.  This bug report should be closed :)

closed 2017-02-02 by tzapu

DaleMitchell, 2018-12-21:
I disagree. The purpose of WiFiManager is to help a user connect to their WiFi network. Part of connecting is getting an IP address. Maybe this proposed feature of displaying the IP address obtained by DHCP is useful to a minority of issues, but it is still a good feature for some including me. I think this is worth pursuing. At the moment, it appears that the page displayed after saving WiFi credentials has a redirect to the `/i` page, so if the wireless device fails to connect to the AP (usually due to incorrect credentials), the user will get redirected to the `/i` page where it shows the local IP of the wireless device, which at the moment happens to be the IP for accessing the main page of an instance of WiFiManager. Why does it not redirect to a page such as `Connecting to the ESP failed, redirecting to the main menu...` so then the user can retry? It seems like someone was trying to get the info page to actually show the local IP obtained from DHCP after connecting to the desired AP. Hopefully someone can beat me to a pull request, this would be a nice feature.

tablatronix, 2018-12-21:
after connecting to the desired ap, there is no portal , so what is the point, the softap closes when save succeeds. you are talking about client refresh and checking , the server is not available when connecting to a saved wifi, there are many issues to work around to get a clean solution to that

DaleMitchell, 2018-12-21:
You're correct. The SoftAP closes after the ESP successfully connects. Yes, the server is not available when connecting to the user's saved SSID, that is a remarkable challenge. But that doesn't mean it can't reopen the SoftAP for a brief period of time to report back to the user, and then once the ESP acknowledges the user received an updated info page, then it can close the SoftAP and reconnect to the user's saved WiFi SSID. Remember that DHCP assigns IP addresses with a lease, which should always be plenty more than sufficient time for retaining the same IP address.

tablatronix, 2018-12-21:
Then we have to add js checking, captiveportals might close down on the user, dropped ap connections can be a problem if reconecting to fast regardless. And then you are assuming the ip is the same. Its on the roadmap. 

Ideally we connect to the router in sta+ap mode then there is no problem, but that is unstable and can be a problem, so it needs to address all this and have good design or else it will introduce bugs.