tzapu/WiFiManager #266: Cannot automatically reconnect after router power cycle

reynolga, 2016-12-17:
The wifi manager works well for the most part. I've encountered a strange issue though. If the router is reset while a device is connected to it, the device is unable to remember the credentials previously used. The device must be restarted and it will set up an Autoconnect network to configure the SSID and password. I haven't been able to solve this. Any ideas?

cosmocracy, 2017-01-03:
How do you know that the device has "forgotten" the credentials and just can't reconnect? I'm troubleshooting something similar and was just wondering how you've been able to troubleshoot this.  Thanks!

reynolga, 2017-01-03:
So the device seems to be stuck after the router power cycle. I don't detect the device on my server and it does not set up an access point. I've also had it connected to my computer during this process and I don't see any serial printouts after the router power cycle,

To fix this I have I go to the ESP8266 device and disconnect it and reset the power. After the ESP8266 powers back up, it will set up a webserver - "Autoconnect" which sets up when it can't connect with other credentials. So that is why I say it "forgot" the credentials. It has no recollection of being previously connected.

If the ESP device is not powered up during this router power cycle. These will be fine and connect to the network the next time the ESP device powers on. It only affects the devices that are currently connected to the network. 

That's all I've figured so far.. let me know if you find more!!

cosmocracy, 2017-01-03:
@reynolga Thanks for the info. While surfing for ESP8266 and disconnect/reconnect code, I saw so many examples--some of which worked and some that didn't.  I also came across this issue with the ESP8266 Wifi library itself which suggests that the latest version could have some quirks, too: https://github.com/esp8266/Arduino/issues/2174

I've added a suggestion for @tzapu that we somehow encapsulate reconnection code to make it dead-simple for sketch programmers to call within their loop() code.  See issue #272

reynolga, 2017-01-04:
Good work, looks like you spent quite a bit of time troubleshooting this. I'm tracking your issue #272 . Hopefully we can get something working here. Having all of the devices lose connection and have to be manually reset is such a pain.

tablatronix, 2017-02-08:
The esp reconnects by itself, there is no code required in loop. The only thing it does is what your sketch tells it to. if you have autoconnect on startup and it cannot connect, it will start configportal, if you have a timeout set it will exit automatically and continue to retry ( the esp sdk does this internally )

I am not sure how this loss of connection is happening but it is hard to pin in on this library , i have been trying to reproduce.

you guys also do not mention anything you are using, what sdk, what branches?

sabretus, 2017-11-04:
@tablatronix I'm using [this](https://github.com/squix78/esp8266-weather-station/blob/master/examples/WeatherStationDemoExtendedDST/WeatherStationDemoExtendedDST.ino) project.

I have 0.96" screen attached to my ESP. Sometimes when there is a power outage at home, all devices are turned off and on at the same time. WIFI router boot time takes longer than ESP, and then I see the following message on the ESP screen: "Wifi manager, please connect to xxxxx". It behaves like it requires initial setup.

Once power on the ESP reset manually, it's successfully connected to WIFI AP.

tablatronix, 2017-11-04:
Are you using setconfigportaltimout ?
Are you losing wifi credentials also ?

ofekp, 2018-08-18:
To anyone encountering the same issue in the future, this is my wif connection method which works well.
**Before using `delay(1200);` it did not work and stalled on `WiFi.status() != WL_CONNECTED`.**

```
void connectWiFi() {
  WiFi.disconnect();
  delay(1200);
  WiFi.mode(WIFI_STA);
  Serial.println(F("connectWiFi"));
  WiFi.begin(ssid, password);

  // wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(F("."));
  }
  
  Serial.println(F(""));
  Serial.print(F("Connected to "));
  Serial.println(ssid);
  Serial.print(F("IP address: "));
  Serial.println(WiFi.localIP());
}
```

devabhixda, 2020-01-21:
Adding a delay of about 2 min before initializing auto connect should solve the issue as nearly every router would be able to turn on in the given duration after power comes back. Not an issue with WifiManager.

erickfloors, 2021-02-16:
> Adding a delay of about 2 min before initializing auto connect should solve the issue as nearly every router would be able to turn on in the given duration after power comes back. Not an issue with WifiManager.

which one Adding a delay of about 2 min before initializing auto connect ? can you tell me..which part should I add

devabhixda, 2021-02-16:
> which one Adding a delay of about 2 min before initializing auto connect ? can you tell me..which part should I add

  `WiFiManager wifiManager;
  //Delay
  delay(300000);
  wifiManager.autoConnect();`

erickfloors, 2021-02-16:
> `WiFiManager wifiManager;`
> //Delay
> `delay(300000);`
> `wifiManager.autoConnect();`

my code like this..

if (!wifiManager.autoConnect("SmartHome", "password123")) {
    Serial.println("Failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }

which part should I add ?  Thanks You

devabhixda, 2021-02-16:
> my code like this..
>
> if (!wifiManager.autoConnect("SmartHome", "password123")) {
> Serial.println("Failed to connect and hit timeout");
> delay(3000);
> //reset and try again, or maybe put it to deep [...]

Before the autoconnect block

erickfloors, 2021-02-16:
> autoConnect

WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 33);   // was 32 length
 
  Serial.println(blynk_token);
 
  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;
 
  wifiManager.setSaveConfigCallback(saveConfigCallback);   //set config save notify callback
 
  //set static ip
  // this is for connecting to Office router not GargoyleTest but it can be changed in AP mode at 192.168.4.1
  //wifiManager.setSTAStaticIPConfig(IPAddress(192,168,10,111), IPAddress(192,168,10,90), IPAddress(255,255,255,0));
 
  wifiManager.addParameter(&custom_blynk_token);   //add all your parameters here
 
  //wifiManager.resetSettings();  //reset settings - for testing
 
  //set minimu quality of signal so it ignores AP's under that quality
  //defaults to 8%
  //wifiManager.setMinimumSignalQuality();
 
  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep, in seconds
  wifiManager.setTimeout(600);
 
  //fetches ssid and pass and tries to connect, if it does not connect it starts an access point with the specified name
  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect("SmartHome", "password123")) {
    Serial.println("Failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }
 
  //if you get here you have connected to the WiFi
  Serial.println("Smart  Home is Connected");   
 

can you tell me or give a sample with code above? please

erickfloors, 2021-02-18:
> Before the autoconnect block

can you give a sample on my code on below?

devabhixda, 2021-02-18:
> WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 33); // was 32 length
>
> Serial.println(blynk_token);
>
> //WiFiManager
> //Local intialization. Once its business is done, [...]

https://del.dog/phycogaper.txt

erickfloors, 2021-02-18:
> https://del.dog/phycogaper.txt

thanks you very much..i will try