tzapu/WiFiManager #267: What is the supposed behaviour for this use case?

neilyoung, 2016-12-26:
The setup:

  ```
void setup() {
     Serial.setDebugOutput(true);
     WiFiManager wifiManager;
     wifiManager.setTimeout(120);
     wifiManager.setConnectTimeout(30);
     if (!wifiManager.autoConnect("Autoconnect")) {
        Serial.println("ESP Reset");
        ESP.restart();
    }
    Serial.println("connected...yeey :)");
}
```

I'm starting with last known AP disabled. As expected a couple of seconds later the Autoconnect network appears. I'm doing nothing, just re-enabling the last known AP.

Before that the log was full with things like this:

```
no HOME-G found, reconnect after 1s
reconnect
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (29)
reconnect

```

Finally after re-enabling HOME-G this log entry appears:

```
connected with HOME-G, channel 6
dhcp client start...
ip:192.168.179.20,mask:255.255.255.0,gw:192.168.179.1
```

and a lot more later this:

`connected...yeey :)
`

So now I was expecting to not see the Autoconnect network anymore, but it is still there, wasting power. 

a) What can I do in order to switch the Autoconnect AP off
b) Is this the expected behavior?

Regards

tzapu, 2017-01-02:
the network left there might be a bug of how it enables AP_STA mode.

should be sorted whenever i get around to doing more changes to the dev branch

cheers

tablatronix, 2017-02-08:
configportal unload does not always reset mode
notably timeout

most people do not know that ap mode is also persistent ( this was a surprise to me as it is not well documented anywhere. )

add a ` WiFi.mode(WIFI_STA);` before your esp.reset as a workaround.

tablatronix, 2017-02-08:
@tzapu want me to pr this ?

tzapu, 2017-02-08:
in examples or part of the lib?

it's a surprise to me as well that _AP_STA is persistent, but i guess that explains some weird behaviour as well

cheers

tablatronix, 2017-02-08:
#312 

I will look into the ask and see if there is an option for this, but I think there is only 1 persistent toggle.

closed 2018-01-26 by tablatronix