tzapu/WiFiManager #278: Wiring diagram

Naythan, 2017-01-06:
Can we add a link to how to wire the esp8266-01 to the arduino.  Make it simpler for a newb like myself to start with using the esp8266-01.
Thank you

pieman64, 2017-01-06:
@Naythan what does your post have to do with the WiFiManager?

Naythan, 2017-01-06:
WiFiManager works with the ESP3266.  For someone who isn't familiar with adding code to the ESP, such as myself, it would be handy to know how to hook it up to the Arduino, or if it should be connected to something else.  It's kind of the first step to using it for projects.  All that's needed is a link to follow.

scropion86, 2017-01-07:
@Naythan this library works with ESP8266 boards standalone means all the code runs on the ESP8266 chip
and not as attached serial communicator to main Arduino board like UNO for example.

alx-1, 2017-01-07:
Get a huzzah board and follow the instructions from adafruit...it's a good place to start.

closed 2017-01-08 by Naythan