tzapu/WiFiManager #279: On Demand Portal SSID with ChipID?

simonliu009, 2017-01-07:
Hi tzapu,
The on demand portal is working like a charm.
Right now I can only use fixed SSID for the wifiManager.startConfigPortal(), is there any way to add the ChipID to the on demand portal SSID like the wifiManager.autoConnect() does?
I tried the following code but it fails to compile with the "no matching function for call to 'WiFiManager::startConfigPortal(String&)'  wifiManager.startConfigPortal(onCmdSSID);" error .
          `           if ( digitalRead(apConfig) == LOW ) {
              Serial1.print("Entering config mode on demand...");
              // WiFiManager wifiManager;
              String onCmdSSID = "ESP" + String(ESP.getChipId());
              wifiManager.startConfigPortal(onCmdSSID);
              Serial1.println("connected...yeey :)");
            }`

tzapu, 2017-01-10:
hi,

it s been added to the development version
https://github.com/tzapu/WiFiManager/tree/development

cheers

ArnieX, 2017-02-12:
Hi @simonliu009,

I do it this way and it works:

```arduino
// Define variable
String autoconf_ssid = "SWIFITCH_"+String(ESP.getChipId());
// Later calling autoconnect
wifiManager.autoConnect(autoconf_ssid.c_str());
```

Hope this helps.

closed 2017-08-27 by tablatronix