tzapu/WiFiManager #280: Calling WiFi.disconnect() seems to wipe WiFiManagers SSID/Password settings?

horendus, 2017-01-07:
Hey

I have been using WiFiManager for a project, fantastic library!

I have however run into a problem, my project needs to be able to connect/disconnect wifi at will.

Connecting is fine using wifiManager.autoConnect(); however if I call WiFi.disconnect(); at any point I cannot re connect to my wifi using wifiManager.autoConnect(); unless I go through the portal again and re enter my wifi detail.

The only think I can think of is WiFi.disconnect(); is wiping the saved wifi credentials for the library.

Is there another way I should be disconnecting from wifi? WiFiManager doesnt have a disconnect function.

Thanks for any help you can provide!

ortegafernando, 2017-01-12:
See this: https://github.com/tzapu/WiFiManager/issues/242

It is about persistent(false)

tablatronix, 2017-01-12:
```
/**
 * Disconnect from the network
 * @param wifioff
 * @return  one value of wl_status_t enum
 */
bool ESP8266WiFiSTAClass::disconnect(bool wifioff) {
    bool ret;
    struct station_config conf;
    *conf.ssid = 0;
    *conf.password = 0;

    ETS_UART_INTR_DISABLE();
    if(WiFi._persistent) {
        wifi_station_set_config(&conf);
    } else {
        wifi_station_set_config_current(&conf);
    }
    ret = wifi_station_disconnect();
    ETS_UART_INTR_ENABLE();

    if(wifioff) {
        WiFi.enableSTA(false);
    }

    return ret;
}
```

ortegafernando, 2017-01-13:
From this, for me is clear (almost ;) : https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/generic-class.md
that explain this:

"""""
 Module is able to reconnect to last used Wi-Fi network on power up or reset basing on settings stored in specific sectors of flash memory. By default these settings are written to flash each time they are used in functions like WiFi.begin(ssid, password). This happens no matter if SSID or password has been actually changed.
 This might result in some wear of flash memory depending on how often such functions are called.
 Setting persistent to false will get SSID / password written to flash only if currently used values do not match what is already stored in flash.
 Please note that functions **WiFi.disconnect** or WiFi.softAPdisconnect reset currently used SSID / password. If persistent is set to false, then using these functions will not affect SSID / password stored in flash.
 To learn more about this functionality, and why it has been introduced, check issue report #1054
 """"

Meanwhile I have forked @kentaylor library and added option to use persistent(false), I will try to use it and see if there is any problema (I have still not use it)

I think will be the best to have persistent = false as default value in this library.

Regards

tablatronix, 2017-01-13:
I suppose the question is why is this library calling disconnect to begin with?

azevedo-manuel, 2018-01-04:
Yes, I confirm this and the workaround by @tablatronix works with a minor change:

```
WiFi.persistent(false);      
WiFi.disconnect();          
WiFi.persistent(true);
```

Also answerting @tablatronix, I call this function when a user presses a button to call Wi-Fi Manager. If the ESP8266 is still connected to a network, without `WiFi.disconnect()` it would still be connected to that network and it would serve the Wi-Fi Manager webpage if a user connected to it.

closed 2018-01-26 by tablatronix