tzapu/WiFiManager #291: Newbie issue I hope.

zmeu1975, 2017-01-11:
I have been writing some code for the esp8266 but it was messy so I decided to use your WiFi manager library. It run through the autoconect and connected to my WiFi router. It gave me the IP 192.168.8.104. But when I connect to that it says page can not be displayed. I tried to ping it but nothing. OK I thought and went back to my original working but messy sketch and now that say the same thing too. 
I then uploaded the example webserver for the ESP and again the same result. What am I missing?

tzapu, 2017-01-12:
hi, you might have some flash corruption although i haven t seen that ip pop up until now

try a WiFi.disconnect()
you might also try flash erase config from the sdk
check kentaylor's flash erase and see if it helps

 cheers

scropion86, 2017-01-12:
@zmeu1975 here is [how to erase the flash](https://github.com/kentaylor/EraseEsp8266Flash)

zmeu1975, 2017-01-13:
@scropion86 Thank you for that. However every time run it it stack dumps at address 64.

j = 63
addressSucceed: 258048 - 0x3f000

Exception (0):
epc1=0x40210818 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffca0 end: 3fffffb0 offset: 01a0

zmeu1975, 2017-01-13:
Thank you guys. Although that did not work it did set me on the right track. In the end I used the esptool to erase the firmware. All working now.

esptool.py --port /dev/ttyUSB1 erase_flash

@scropion86 @tzapu

closed 2017-01-13 by zmeu1975