tzapu/WiFiManager #294: Where Does This Text Come From?

TSJim, 2017-01-13:
First, I really want to thank everyone who has worked so hard to make this library available to us!

Here's my question:  After I select the AP's SSID from my phone or computer's WiFi network settings, and I use the AP mode configuration screen to select my WiFi network SSID and enter the password and click "Save", the exact following text appears:

```
Credentials Saved
Trying to connect ESP to network.
If it fails reconnect to AP to try again
```

I cannot for the life of me figure out where this text comes from!

I've searched my entire computer (Mac), looked through every file manually in the WiFiManager distribution including in the HTML template, and I've done a Google search for the terms in the message, and I can not find it ANYWHERE!

I'm probably being really dumb, but can someone please point me to where this text lives?

Thank you so much!

-- Jim

tzapu, 2017-01-13:
Hi

thanks for using it

see HTTP_SAVED in WiFiManager.h

cheers

TSJim, 2017-01-13:
Wow, I never would have found that!  Your pointer led me to exactly what I needed in order to make the needed changes.

Does that get text saved into the ESP's flash at the factory?

Thank you so much for your fast reply!

tzapu, 2017-01-13:
no, this is just part of the library

cheers

TSJim, 2017-01-13:
Hello!

You're a night person too...either that, or you aren't in the US.  ;)

Wow, I just don't see where that text gets written to flash anywhere in your library files.  Is it in your library, or one of the others used for ESP8266 WiFi?  Just curious.  I see the text in the latest version of the "extras/WiFiManager.template.html" file (I had an older one I guess), but if I change the text in that file, it does not change in the AP's "Credentials Saved" text message.

I wrote a quick test sketch (below) to just read the flash variable "HTTP_SAVED" that you referred me to, and it displays the original text just fine.  I just cannot understand how that text got into the flash memory from your library.

```
#include "WiFiManager.h"

void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
String test = FPSTR(HTTP_SAVED);
Serial.println(test);

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Sorry for not knowing this stuff.  I envy your programming skills!  I've been using Arduinos since they came out, and have done lots of programming before and after that.  I'm just new...as many people are...to the ESP8266 family.  There seem to be lots of hidden knowledge.

I have a question about a programmatic reset after the credentials are saved not working (which forces me to have to press the physical reset button that I added), but I'll post that in a new thread.  Maybe you can lend your expertise on that thread later as well...that would be great!

I wish GitHub had a "DONATE TO THE PROJECT" button, because I would use it right now!

Thanks!

jpaustin, 2017-01-20:
I believe the compiler sees the string type and will allocate flash for the data. At run time the flash string is retrieved and stored in RAM that is used for the actual setup and loop sections to call upon.

closed 2017-08-27 by tablatronix