tzapu/WiFiManager #296: WiFiManager Captive portal not showing up on my iOS devices

kirhim, 2017-01-17:
Hi, I am trying to make an IoT mood lamp with 8266. I am using Firebase server so that I could control the device from anywhere. Also, I have created an iOS app to control(turn on/off the lights) the device. 

The problem I am having is this: 
Say, I need to connect my device to a new network(wifi) but the device is not configured to the new network. In this case, I could just simply update the new network(SSID, PASSWORD) to Arduino IDE. But I am doing this project for commercial purposes and I want to provide simple and easy service to my customers. So I just want to connect any new network using my iOS app or in a simple way (without any coding). WiFiManager would make this possible right?

I was so glad that I found your code it was a real time saver for me. 
The problem I am having is that even when I run the Arduino IDE code without any errors,  I can't get the captive portal open in my iPhone and Mac Pro. 
I guess I am suppose to generate a wifi network on my iOS device (ESP + ChipID) However, I don't see it. How do I get the view "How it looks" in the README.md file?  

I've been trying to solve this issue for the past few days but still haven't found the right answer. 
Please let me know if I am misunderstanding anything. Any comments would be very appreciate it. 

Thanks,

Ki

MarcFinns, 2017-02-06:
behaviour confirmed but it happens randomly.
have you managed to get a solution?

tablatronix, 2017-02-08:
click ap, click forget network.

Stibbons, 2017-03-10:
Got this issue today. Forgetting the wifi network is enough, next try worked. Occured on some weird hostel connection, should be a bug from apple.

andysc, 2017-04-02:
I'm seeing a similar problem - Mac OS (Sierra) says "The Wi-Fi network "GlowOrb-1234-5678" (the name I gave it) could not be joined". Sometimes it works. Other times not.
When it does work, and the captive portal window pops up, it sometimes loads the home page and works fine, and other times says "a problem occurred. The web page couldn't be loaded". 

Debug is not showing any problem:
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Handle root

pieman64, 2017-04-02:
@andysc have you tried changing the SSID to simply GlowOrb ?

tablatronix, 2017-04-02:
Was the device previusly joined or had credentials already to a non existant ap?

MarcFinns, 2017-04-02:
I confirm that with Mac and iOS it is very difficult to connect. It tends to immediately revert to the wifi it was connected to in the first place. In some cases i had to switch off the home router to make the ESP the only wifi and get the connection. Also, in many cases i have to manually open a browser and go to 192.168.4.1 otherwise the config page does not come up...

andysc, 2017-04-02:
@pieman64 no - was there a reason for suggesting that? (length, characters)?

andysc, 2017-04-02:
@tablatronix yes, it was previously programmed (using normal ESP wifi stuff) to connect to a different AP

andysc, 2017-04-02:
changed AP to GlowOrb ... it didn't show up in the list of available access points until I turned wifi off then on, then it saw it. Spent a long time trying to connect to it (with the wifi logo going up and down)... before finally connecting, and doing the captive portal thing properly. 
Oooh, are we onto something here @pieman64, or was that just lucky ;)

andysc, 2017-04-02:
It also seems that once it has failed to find the AP that was configured (i.e. if you turn the access point off), and it goes into config mode, if you don't set up a new association, it forgets the old AP settings and makes hardly any attempt to connect to them if you power cycle the ESP. 
Thus, if your network was down for a while and you turned the ESP on, it would fail to find your AP and then after that would require reconfiguration before being able to connect again.
Or is there a parameter to fix that somewhere?

andysc, 2017-04-02:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: SET AP STA
Entered config mode
192.168.4.1
GlowOrb
*WM: 
*WM: Configuring access point... 
*WM: GlowOrb
*WM: AP IP address: 

... note the connection result 0, which comes up immediately.

Is there a way to find out what AP the ESP is hunting for?

tablatronix, 2017-04-02:
WiFi.printDiag(Serial)

andysc, 2017-04-02:
aha :) thanks - I also found:
  Serial.print("Looking for ");
   Serial.print(WiFi.SSID().c_str());
   Serial.print(" / ");
   Serial.println(WiFi.psk().c_str());
  
If you get to the captive portal landing page and click Configure WiFi - that seems to be the point it erases the old SSID and password. 
They're coming up as blank in my debug.

andysc, 2017-04-02:
OK - so my point about power failures was not valid... as long as you don't go into the configuration page, your settings remain intact.

andysc, 2017-04-02:
I've gone back to the longer SSID now (GlowOrb-1234-5678) and it seems to connect after a couple of tries, so I don't think the length was significant. SSIDs should be OK up to 32 chars, according to the spec. Whether all the ESP and library buffers are up to that, I don't know :)

tablatronix, 2017-04-02:
There are several known bugs these are both known. I cannot recall the loss of credentials one at this time.

andysc, 2017-04-02:
@tablatronix OK - thanks - the loss of credentials is not as I suggested - it's only if you go to the wifi config page but don't select or save any new credentials. I think that is a bug, but not as bad as I thought ;) Thanks for your help!

andysc, 2017-04-02:
OK, I've found the problem with the wifi credentials getting deleted when you go into the Wifi config page... 
in WifiManager.cpp, handleWifi calls WiFi.scanNetworks() to scan for networks.
in ESP8266WiFiScan.cpp, scanNetworks checks the current connection status, and if it is not STATION_GOT_IP or STATION_IDLE, then it calls WiFi.disconnect(false);
If the ESP can't find the AP it's looking for (which is what causes it to enter AP config mode in the first place), the status will be STATION_NO_AP_FOUND (3), so it will call disconnect.

WiFi.disconnect() deletes the currently stored SSID and password.

Now, whether the correct fix is to look at other options before calling disconnect, or whether calling disconnect is the wrong thing to do at all, I don't know, but I commented out the call to WiFi.disconnect(false), and it all works fine :)

Now the credentials don't get wiped if you go to the AP config page but don't save a new AP.

I'll leave others to work out the best fix, but I'm happy with this for now :)

andysc, 2017-04-02:
BTW, that bit in connectWifi, where it says
*WM: Connection result:
*WM: 0
i.e. 
  int connRes = waitForConnectResult();
  DEBUG_WM ("Connection result: ");
  DEBUG_WM ( connRes );

is greatly improved by adding this bit of code:

  // interpret connRes code
  switch(connRes) {
        case WL_NO_SHIELD:
                DEBUG_WM ("NO SHIELD");
                break;
        case WL_IDLE_STATUS:
                DEBUG_WM ("IDLE STATUS");
                break;
        case WL_NO_SSID_AVAIL:
                DEBUG_WM ("NO SSID AVAILABLE");
                break;
        case WL_SCAN_COMPLETED:
                DEBUG_WM ("SCAN COMPLETED");
                break;
        case WL_CONNECTED:
                DEBUG_WM ("CONNECTED");
                break;
        case WL_CONNECT_FAILED:
                DEBUG_WM ("CONNECT FAILED");
                break;
        case WL_CONNECTION_LOST:
                DEBUG_WM ("CONNECTION LOST");
                break;
        case WL_DISCONNECTED:
                DEBUG_WM ("DISCONNECTED");
                break;
  }

Sorry, I don't yet know how to do a pull request, so someone else will have to put it in if you want this ;)

andysc, 2017-04-03:
I've made 3 small changes, which seem to help a lot... 
Again, I'm sorry I don't know how to do a pull request, but here are the changes...

1. in startConfigPortal
use pure AP mode rather than AP_STA mode (@tablatronix suggested this too in his PR)
  //WiFi.mode(WIFI_AP_STA);
  //DEBUG_WM("SET AP STA");
  WiFi.mode(WIFI_AP);
  DEBUG_WM("SET AP");

2. also in startConfigPortal, go into STA mode before trying to connect with new credentials...

      WiFi.mode(WIFI_STA);
      DEBUG_WM("SET STA");

      // using user-provided  _ssid, _pass in place of system-stored ssid and pass
      if (connectWifi(_ssid, _pass) != WL_CONNECTED) {
        DEBUG_WM(F("Failed to connect."));
      } else {
        //connected
        //WiFi.mode(WIFI_STA);

3. in connectWifi, do a wifi_station_disconnect before trying to begin wifi with new credentials, as we do before trying with old credentials.

  if (ssid != "") {
      //trying to fix connection in progress hanging
      ETS_UART_INTR_DISABLE();
      wifi_station_disconnect();
      ETS_UART_INTR_ENABLE();

    WiFi.begin(ssid.c_str(), pass.c_str());
  } else {

This has made connecting to the AP and bringing up the captive portal far more reliable on a Mac, and also made it more reliable at connecting to the new network when you've configured it.

MarcFinns, 2017-04-03:
You should:
1) clone the repository so that you have your own private copy (1 click action)
2) update the files in your repo as above
3) submit it as pull request to the author so that he can pull the changes into the main branch

andysc, 2017-04-03:
@MarcFinns thank you :)

tzapu, 2017-04-03:
HI Andy, thanks, if you try to make the pull request, make sure you start and submit it to the development branch

cheers
alex

MarcFinns, 2017-04-03:
@tzapu-good to hear you are still evolving it!

andysc, 2017-04-03:
GitHub n00b time again - sorry...
I've changed the files. Done a 
```
git commit -a
```
Not sure what to do next. I *think* it's a 
```
git push
```
but I'm getting
```
remote: Permission to tzapu/WiFiManager.git denied to andysc.
fatal: unable to access 'https://github.com/tzapu/WiFiManager.git/': The requested URL returned error: 403
```
maybe git push needs some parameters?

MarcFinns, 2017-04-04:
Easiest way is to drag and drop the files via website...

andysc, 2017-04-04:
@MarcFinns I think your 3 step instructions missed out "click fork"...?
I'm getting the hang of this slowly. 
Can't drag/drop files as I'm not authorised

MarcFinns, 2017-04-04:
You are authorised in your own rep!! (Not the original that you cloned)

Stibbons, 2017-04-04:
You can find a lot of documentation online about the github workflow. I [like this one](https://gist.github.com/Chaser324/ce0505fbed06b947d962).

andysc, 2017-04-05:
Thanks @Stibbons that was quite helpful (though missed out some crucial steps!).
I now have a PR [https://github.com/tzapu/WiFiManager/pull/349]
Thank you all for your patience :)

closed 2018-01-26 by tablatronix