tzapu/WiFiManager #299: Is there documentation on Connection Result variables

mfzeidan, 2017-01-21:
I'm having trouble finding anything where I can read about what the connection results (0-4 I believe) mean. They seem to change from 0, 3 or 4 and i want to set deepsleep options based on these connection results.

RuBiCK, 2017-02-17:
I'm getting connection result 4 and I don't know what does it mean so I cannot make any troubleshooting because I don't know if it's a password/cipher or any other thing

closed 2017-08-27 by tablatronix

witnessmenow, 2017-10-24:
Why was this closed? Is there documentation on this?

mfzeidan, 2017-10-24:
No idea, I never got an answer

tablatronix, 2017-10-24:
```
String WiFiManager::getConnResString(uint8_t status){
    switch(status) {
        case STATION_GOT_IP:
            return "WL_CONNECTED";
        case STATION_NO_AP_FOUND:
            return "WL_NO_SSID_AVAIL";
        case STATION_CONNECT_FAIL:
        case STATION_WRONG_PASSWORD:
            return "WL_CONNECT_FAILED";
        case STATION_IDLE:
            return "WL_IDLE_STATUS";
        default:
            return "WL_DISCONNECTED";
    }
}
```

tablatronix, 2017-10-27:
That is the conn result for the SDK
It should be the wl_status for WiFi.status()

```
String WiFiManager::getWLStatusString(uint8_t status){
  const char * const WIFI_STA_STATUS[]
  {
      "WL_IDLE_STATUS",
      "WL_NO_SSID_AVAIL",
      "WL_SCAN_COMPLETED",
      "WL_CONNECTED",
      "WL_CONNECT_FAILED",
      "WL_CONNECTION_LOST",
      "WL_DISCONNECTED"
  };

  if(status >=0 && status <=6) return WIFI_STA_STATUS[status];
  return "UNKNOWN";
}
```